ros2/rmw_connextdds #2: Add branch that supports post-Foxy releases (Rolling)

spiderkeys, 2020-12-08:
I've created a branch that removes the structs and enum that are now duplicately defined in ros2/rmw such that this RMW implementation can be used in ROS2 Rolling.

https://github.com/spiderkeys/rmw_connextdds/tree/rolling

I've tested this change with a non-trivial set of applications using ROS2 Rolling (built as of today) on Ubuntu 20.04 and using RTI Connext Pro 6.0.1.

asorbini, 2020-12-08:
Hi Charles,

thank you for your contribution, but unfortunately I think that your work might be a duplicate of the `master` branch of the repository, which is meant to track ROS2 Rolling.
Did you give that branch a try? By inspecting your repository I see that you branched from `foxy`.

The `master` branch is the main development branch, and all other branches share (almost) the same source code, differing only in the default value set for `RMW_CONNEXT_RELEASE` in [static_config.hpp](https://github.com/rticommunity/rmw_connextdds/blob/master/rmw_connextdds_common/include/rmw_connextdds/static_config.hpp#L128) (the `dashing` branch also includes some differences in `package.xml` since some package dependencies - namely `rmw_dds_common` - are missing from that release).

Since the only difference is that symbol definition, the `foxy` branch is mostly provided for convenience, and it should be possible to also build the `master` branch with ROS2 Foxy by passing `--cmake-args -DRMW_CONNEXT_RELEASE=foxy` to `colcon`.
There is also [code in CMakeLists.txt](https://github.com/rticommunity/rmw_connextdds/blob/master/rmw_connextdds_common/CMakeLists.txt#L124) that tries to guess the current ROS version from the environment but I'm guessing that's not quite working yet.

Actually, by reviewing the files just now, I noticed the possible root cause for this: `rmw_connextdds_common` [sets the definition as private](https://github.com/rticommunity/rmw_connextdds/blob/master/rmw_connextdds_common/CMakeLists.txt#L74) when building the "common" library, which probably prevents it from propagating to `rmw_connextdds`.

Anyway, please review the `master` branch, and let me know if you run into any issues or if you see any incompatibilities that should be resolved to support the Rolling release.

spiderkeys, 2020-12-08:
Hi @asorbini , thanks for the update! 

Sorry, I hadn't realized that master was a working rolling branch. I had been working with the foxy branch for the last month and saw that the foxy branch was 133 commits ahead and 137 commits behind master, but didn't know what the working state of master was. I applied the smallest delta required to make the foxy branch work on my rolling build and everything seemed ok :sweat_smile: 

I'll try the master branch moving forward and let you know if I run into any issues!

closed 2020-12-08 by spiderkeys