ros2/rmw_connextdds #3: Could you point me where to find "resource" files, e.g. "em" files?

rockcdr, 2020-12-10:
For example, original one is, https://github.com/ros2/rosidl_typesupport_connext/blob/dashing/rosidl_typesupport_connext_cpp/resource/msg__type_support.cpp.em

asorbini, 2020-12-11:
Hi @rockcdr,

this new RMW does not perform any code generation during the build, so you won't find any template files in the repository.

Instead, the RMW uses a [custom type plugin](https://github.com/rticommunity/rmw_connextdds/blob/master/rmw_connextdds_common/src/ndds/rmw_type_support_ndds.cpp) which wraps the type support generated by `rosidl_typesupport_fastrtps`. This allows Connext DDS to serialize/deserialize ROS messages directly, without the need for the extra copies that were previously needed to convert between ROS and DDS memory representations.

May I ask you what was your goal if you found the template files?
Maybe I can suggest some alternative approaches.

rockcdr, 2020-12-11:
Thanks for your quick reply and explanation. It is really new version of design, i.e., no code generation. 

I am reading the new version of implementation source code, comparing with the previous version, i.e., `rmw_connext_cpp`, and wondering whether you will remove the dependencies on `fastrtps` and `rosidl_typesupport_fastrtps` in future. From my point of view, FYI, main concern is the dependencies. As you know, especially, when using RTI DDS Micro, usually want to have minimum dependencies.

rockcdr, 2020-12-11:
BTW, do you think it might be easy to add a switch (e.g. in `static_config.hpp`) to select to use `rosidl_typesupport_fastrtps` or use a new version of `rosidl_typesupport_connext`?

asorbini, 2020-12-11:
> I am reading the new version of implementation source code, comparing with the previous version, i.e., `rmw_connext_cpp`, and wondering whether you will remove the dependencies on `fastrtps` and [...]

`rmw_connextdds` does not depend on `fastrtps` as a runtime dependency, but only on the `fastcdr` library (used by `rosidl_typesupport_fastrtps`). `fastrtps` should only be brought in as a build dependency (by `rosidl_typesupport_fastrtps`'s `package.xml`).

We currently don't have plans to replace this dependency with Connext-specific code generation, but it could be an option to explore for the future if there is sufficient interest.

asorbini, 2020-12-11:
> BTW, do you think it might be easy to add a switch (e.g. in `static_config.hpp`) to select to use `rosidl_typesupport_fastrtps` or use a new version of `rosidl_typesupport_connext`?

It could be possibly, but it would probably require some pretty heavy `#ifdef`ing, and introduce dependencies in `package.xml` which cannot be made conditional and would require a dedicated branch.

Actually, one of the main reasons for creating this new implementation was to get rid of the bad performance provided by `rosidl_typesupport_connext`.

The main problem though is that this new RMW uses Connext's C API, so the type plugins generated by that package cannot be used directly. The package would have to be changed to run `rtiddsgen` for C, rather than C++.

Beside removing the external dependency from `fastcdr`, are there any other technical goals that you are trying to achieve by using `rosidl_typesupport_connext`?

rockcdr, 2020-12-12:
Thanks. I am considering to use the new version rmw to evaluate RTI DDS Micro. It seems like I should use `rosidl_typesupport_fastrtps` and `fastcdr`, as a quick solution.

asorbini, 2020-12-12:
Closing this issue for now. Feel free to open new ones to discuss Micro support in more detail.

closed 2020-12-12 by asorbini