ros2/rmw_connextdds #4: Problem compiling on Jetson TX2

SBarzz, 2021-01-21:
**System Info**

- _OS_
  - 18.04 (Jetson TX2)
- *ROS version and installation type*
  - eloquent
- *RTI Connext DDS version and installation type*
  -6.0.1
- *RMW version or commit hash*
  -dashing

**Bug Description**

Trying to compile your repo, following your instructions, but I'm getting this errors:

```
CMake Warning at /opt/rti_connext_dds-6.0.1/resource/cmake/FindRTIConnextDDS.cmake:458 (message):
  CONNEXTDDS_ARCH not specified.  Please set -DCONNEXTDDS_ARCH= to specify
  your RTI Connext DDS architecture
```

So I compiled it using:
`colcon build --symlink-install --cmake-args '-DCONNEXTDDS_ARCH=armv8Linux4.4gcc5.4.0'`

Then I got:
```
Starting >>> rmw_connextddsmicro
--- stderr: rmw_connextddsmicro                                                                                 
CMake Warning at CMakeLists.txt:28 (message):
  Common RMW implementation not found.  Skipping rmw_connextddsmicro.

---
```

And when I tried to:
RMW_IMPLEMENTATION=rmw_connextdds ros2 run demo_nodes_cpp talker
 
I got:
`[ERROR] [rcl]: Error getting RMW implementation identifier / RMW implementation not installed (expected identifier of 'rmw_connextdds'), with error message 'failed to find shared library of rmw implementation. Searched rmw_connextdds, at /tmp/binarydeb/ros-eloquent-rmw-implementation-0.8.2/src/functions.cpp:130', exiting with 1.

`

**Expected Behavior**

Support connext RMW

**How to Reproduce**

Follow the README insturctions on Jetson TX2

```

```

asorbini, 2021-01-21:
Hi @SBarzz,

thank you for your report.

Based on the error message, and since you are not mentioning any actual build failure (the message about rmw_connextddsmicro is a benign warning), I wonder if by any chance you missed to `source install/setup.bash` after running `colcon build`. I often run into the error you experienced when I forget to do so.

Can you please double check that you loaded the `install/setup.bash` script generated by the RMW?

If you already did that, I would appreciate if you could paste the console output from trying to build the repository on your system. You can generate more output by passing `--event-handlers console_cohesion+` (or `console_direct+`) to `colcon build`.

SBarzz, 2021-01-21:
> Hi @SBarzz,
>
> thank you for your report.
>
> Based on the error message, and since you are not mentioning any actual build failure (the message about rmw_connextddsmicro is a benign warning), I wonder [...]

Now, I'm getting this errors:

user@user:~/ros2_connextdds$ source ~/ros2_connextdds/install/setup.bash
[rti_connext_dds_cmake_module][warning] No RTI Connext DDS Micro installation specified via RTIMEHOME. RTI Connext DDS Micro will not be available unless you already added it to LD_LIBRARY_PATH.
user@user:~/ros2_connextdds$ RMW_IMPLEMENTATION=rmw_connextdds ros2 run demo_nodes_cpp talker

>>> [rcutils|error_handling.c:107] rcutils_set_error_state()
This error state is being overwritten:

  'failed to resolve symbol 'rmw_create_node' in shared library '/home/user/ros2_connextdds/install/rmw_connextdds/lib/librmw_connextdds.so', at /tmp/binarydeb/ros-eloquent-rmw-implementation-0.8.2/src/functions.cpp:163'

with this new error message:

  'failed to resolve symbol 'rmw_create_publisher' in shared library '/home/user/ros2_connextdds/install/rmw_connextdds/lib/librmw_connextdds.so', at /tmp/binarydeb/ros-eloquent-rmw-implementation-0.8.2/src/functions.cpp:163'

rcutils_reset_error() should be called after error handling to avoid this.
<<<

>>> [rcutils|error_handling.c:107] rcutils_set_error_state()
This error state is being overwritten:

  'failed to resolve symbol 'rmw_create_publisher' in shared library '/home/user/ros2_connextdds/install/rmw_connextdds/lib/librmw_connextdds.so', at /tmp/binarydeb/ros-eloquent-rmw-implementation-0.8.2/src/functions.cpp:163'

with this new error message:

  'failed to resolve symbol 'rmw_create_subscription' in shared library '/home/user/ros2_connextdds/install/rmw_connextdds/lib/librmw_connextdds.so', at /tmp/binarydeb/ros-eloquent-rmw-implementation-0.8.2/src/functions.cpp:163'

rcutils_reset_error() should be called after error handling to avoid this.
<<<

>>> [rcutils|error_handling.c:107] rcutils_set_error_state()
This error state is being overwritten:

  'failed to resolve symbol 'rmw_create_subscription' in shared library '/home/user/ros2_connextdds/install/rmw_connextdds/lib/librmw_connextdds.so', at /tmp/binarydeb/ros-eloquent-rmw-implementation-0.8.2/src/functions.cpp:163'

with this new error message:

  'Expected lexeme type (19) not found, search ended at index 43, at /tmp/binarydeb/ros-eloquent-rcl-0.8.5/src/rcl/lexer_lookahead.c:223'

rcutils_reset_error() should be called after error handling to avoid this.
<<<

>>> [rcutils|error_handling.c:107] rcutils_set_error_state()
This error state is being overwritten:

  'failed to resolve symbol 'rmw_create_node' in shared library '/home/user/ros2_connextdds/install/rmw_connextdds/lib/librmw_connextdds.so', at /tmp/binarydeb/ros-eloquent-rmw-implementation-0.8.2/src/functions.cpp:163, at /tmp/binarydeb/ros-eloquent-rcl-0.8.5/src/rcl/node.c:336'

with this new error message:

  'rcl node's rmw handle is invalid, at /tmp/binarydeb/ros-eloquent-rcl-0.8.5/src/rcl/node.c:485'

rcutils_reset_error() should be called after error handling to avoid this.
<<<
[ERROR] [rcl]: Failed to fini publisher for node: 1
terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
  what():  failed to initialize rcl node: rcl node's rmw handle is invalid, at /tmp/binarydeb/ros-eloquent-rcl-0.8.5/src/rcl/node.c:485
user@user:~/ros2_connextdds$ 

This is the build console output :
@user:~/ros2_connextdds$ colcon build --symlink-install --cmake-args '-DCONNEXTDDS_ARCH=armv8Linux4.4gcc5.4.0' --event-handlers console_cohesion+
Starting >>> rti_connext_dds_cmake_module
--- output: rti_connext_dds_cmake_module                             
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
[... 499 lines cut ...]
-- Symlinking: /home/user/ros2_connextdds/install/rmw_connextdds/share/rmw_connextdds/environment/library_path.dsv
-- Symlinking: /home/user/ros2_connextdds/install/rmw_connextdds/share/ament_index/resource_index/package_run_dependencies/rmw_connextdds
-- Symlinking: /home/user/ros2_connextdds/install/rmw_connextdds/share/ament_index/resource_index/parent_prefix_path/rmw_connextdds
-- Symlinking: /home/user/ros2_connextdds/install/rmw_connextdds/share/rmw_connextdds/environment/ament_prefix_path.sh
-- Symlinking: /home/user/ros2_connextdds/install/rmw_connextdds/share/rmw_connextdds/environment/ament_prefix_path.dsv
-- Symlinking: /home/user/ros2_connextdds/install/rmw_connextdds/share/rmw_connextdds/environment/path.sh
-- Symlinking: /home/user/ros2_connextdds/install/rmw_connextdds/share/rmw_connextdds/environment/path.dsv
-- Symlinking: /home/user/ros2_connextdds/install/rmw_connextdds/share/rmw_connextdds/local_setup.bash
-- Symlinking: /home/user/ros2_connextdds/install/rmw_connextdds/share/rmw_connextdds/local_setup.sh
-- Symlinking: /home/user/ros2_connextdds/install/rmw_connextdds/share/rmw_connextdds/local_setup.zsh
-- Symlinking: /home/user/ros2_connextdds/install/rmw_connextdds/share/rmw_connextdds/local_setup.dsv
-- Symlinking: /home/user/ros2_connextdds/install/rmw_connextdds/share/rmw_connextdds/package.dsv
-- Symlinking: /home/user/ros2_connextdds/install/rmw_connextdds/share/ament_index/resource_index/packages/rmw_connextdds
-- Symlinking: /home/user/ros2_connextdds/install/rmw_connextdds/share/rmw_connextdds/cmake/ament_cmake_export_libraries-extras.cmake
[... cut ...]
-- Symlinking: /home/user/ros2_connextdds/install/rmw_connextdds/share/rmw_connextdds/cmake/rmw_connextddsConfig.cmake
-- Symlinking: /home/user/ros2_connextdds/install/rmw_connextdds/share/rmw_connextdds/cmake/rmw_connextddsConfig-version.cmake
-- Symlinking: /home/user/ros2_connextdds/install/rmw_connextdds/share/rmw_connextdds/package.xml
---
Finished <<< rmw_connextdds [17.4s]

Summary: 4 packages finished [1min 35s]
  2 packages had stderr output: rmw_connextdds_common rmw_connextddsmicro
user@user:~/ros2_connextdds$ source ~/ros2_connextdds/install/setup.bash

asorbini, 2021-01-23:
I don't have access to a Jetson TX2 board, but I tried to reproduce this error on a Raspberry Pi 4 using (pseudo) host bundle `rti_connext_dds-6.0.1-pro-armv8Linux4.4gcc5.4.0.tar.gz`.

I had a few warnings from `FindRTIConnextDDS.cmake`, and I had to pass `--cmake-args -DCONNEXTDDS_ARCH=armv8Linux4.4gcc5.4.0 -DRTICONNEXTDDS_VERSION_OK=true` to have it successfully find the installed libraries, but beside that, I was able to successfully compile the RMW and run a hello world with `demo_nodes_cpp/talker` and `demo_nodes_cpp/listener`.

My test was on Ubuntu 20.04, and used ROS foxy. I will try to configure an instance of 18.04 with dashing next.

Since your log didn't include any of the warnings that I received, I'm wondering if I used the correct bundle. Which installer files are you using for your Connext installation?

Also, I've been under the assumption that you are performing a native build on the board, but it would be great if you can confirm that you are not trying to cross-compile the library.

SBarzz, 2021-01-24:
> I don't have access to a Jetson TX2 board, but I tried to reproduce this error on a Raspberry Pi 4 using (pseudo) host bundle `rti_connext_dds-6.0.1-pro-armv8Linux4.4gcc5.4.0.tar.gz`.
>
> I had a few [...]

1.What do you mean by installer files?
I copy the contents of the installed files from "rti_connext_dds-6.0.1-pro-armv8Linux4.4gcc5.4.0.tar.gz" to my TX2 to /opt/rti_connext_dds-6.0.1/.

2. I'm building your branch on my TX2. which means that I run the "colcon build.. " on my TX2.

Thanks

SBarzz, 2021-01-25:
> --cmake-args -DCONNEXTDDS_ARCH=armv8Linux4.4gcc5.4.0 -DRTICONNEXTDDS_VERSION_OK=true

Hi @asorbini, any update?

asorbini, 2021-01-25:
@SBarzz I have identified a couple of issues, and I'm currently working on a fix, that I expect to push later today.

The problem in your case was that we didn't support eloquent, and even though the RMW successfully built with that release, some of the generated symbols (the ones you received errors about) didn't match the expected signature and were thus not properly linked into the final library.

asorbini, 2021-01-26:
The issue should be resolved now. Please let me know if you still run into any problem.

SBarzz, 2021-01-26:
> The issue should be resolved now. Please let me know if you still run into any problem.

Great thanks @asorbini the demo is not working.

But, When I tried to launch the realsense ros node for example with the connext rti RMW, I got this error:

```
[cam1-1] [ERROR] [rmw_connextdds]: introspection type support not found for tf2_msgs::msg::dds_::TFMessage_
[cam1-1] [ERROR] [rmw_connextdds]: failed to generate DDS type code: tf2_msgs::msg::dds_::TFMessage_
[cam1-1] [ERROR] [rmw_connextdds]: failed to register type for writer
[cam1-1] [ERROR] [rmw_connextdds]: failed to allocate RMW_Connext_Publisher
[cam1-1] [ERROR] [rmw_connextdds]: failed to create RMW publisher
[cam1-1] terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
[cam1-1]   what():  could not create publisher: error not set, at /tmp/binarydeb/ros-eloquent-rcl-0.8.5/src/rcl/publisher.c:173
[ERROR] [cam1-1]: process has died
```

Any idea on how to solve it?

asorbini, 2021-01-26:
Thank you for confirming @SBarzz

closed 2021-01-26 by asorbini