ros2/rmw_connextdds #10: How can I use rmw_connextdds in Jetson Nano?(armv8)

ldg810, 2021-03-15:
Hi, I am trying to use RTI DDS with ROS2.

In desktop environment, I succeed with installing RTI DDS 6.0.1 with rmw_connextdds repository.

My working area is drone, so I am trying to install RTI DDS in Jetson Nano Development Kit (armv8 architecture).

I Downloaded "rti_connext_dds-6.0.1-pro-target-armv8Linux4.4gcc5.4.0.rtipkg" and extract it.

There are "rti_connext_dds-6.0.1-pro-target-armv8Linux4.4gcc5.4.0.zip" and I also extracted it into /opt/rti_connext_dds-6.0.1

The difference between amd64 arch and armv8 is that there is no installation process in armv8 version and only pre-built libraries and binaries are there.

When I am trying to build rmw_connextdds after "source rtisetenv_armv8Linux4.4gcc5.4.0.bash" and "export CONNEXTDDS_DIR=${NDDSHOME}", I got following errors.

```
CMake Warning at cmake/rti_build_helper.cmake:572 (find_package):
   By not providing "FindRTIConnextDDS.cmake" in CMAKE_MODULE_PATH this
   project has asked Cmake to find a package configuration file provided by
   "RTIConnextDDS", but CMake did not find one.

   Could not find a package configuration file provided by "RTIConnextDDS"
   (requested version 5.3.1) with any of the follwing names:

       RTIConnextDDSConfig.cmake
       rticonnextdds-config.cmake

   ...
```

Indicating that can not find RTIConnextDDS package.

I found no manual or instructions to use RTI Connext DDS with ROS2 in armv8 architecture.

Is there any suggestion??

ldg810, 2021-03-15:
I did following modifications and I succeed to use RTI connext DDS in Jetson Nano.

- copy include dir, rti_versions.xml and resource/cmake/FindRTIConnextDDS.cmake from desktop RTI 6.0.1 directory to Jetson.
- modify <architecture> in <target_libraries> in rti_versions.xml from "x64Linux2.6gcc4.4.5" to "armv8Linux4.4gcc5.4.0"

Then, everythings works well.

Do you think is this right way to correct the problem..?

asorbini, 2021-03-16:
Hi @ldg810, thank you for your questions!

It looks like you pretty much figured out on your own all of the files required to generate an installation of Connext that can be used for a native build of `rmw_connextdds` on an ARM target.

As you correctly identified, in order to build the repository you will need:
- The `include/` directory containing the header files.
- The `FindRTIConnextDDS.cmake`, and possibly `rti_versions.xml` which that scripts tries to read (although I'm not sure if this is a "hard" requirement or it may be overcome).
- The `lib/<target>` directory containing libraries for the selected build target.
- The `rtisetenv_<architecture>.(bash|bat)` environment scripts (not required, but useful).

Unfortunately, at the moment (6.0.1), Connext does not officially support native builds on ARM/Linux platforms, as libraries for these platforms were often expected to be cross-compiled in the past, so there isn't yet a "host" package for them, like there is, for example, for `x64Linux`, only "targets" (which only contains the binary libraries).

In fact, the file that you used is a "target", and it is meant to be installed on top of an official "host" bundle using `rtipkginstall` rather than extracting its contents manually (see the [official docs](https://community.rti.com/static/documentation/connext-dds/6.0.1/doc/manuals/connext_dds/getting_started/cpp11/before.html#installing-connext-dds) for reference).

In general, in order to install RTI Connext DDS on an ARM target like the Jetson Nano, I recommend following this workflow for now:
1. Install the RTI Connext DDS host package on a Linux x86 system (64-bit or 32-bit) using the official installers, e.g.:
```sh
./rti_connext_dds-6.0.1-pro-host-x64Linux.run
```
2. Install an RTI Connext DDS target package for the target build platform:
```
./rti_connext_dds-6.0.1/rtipkginstall rti_connext_dds-6.0.1-pro-target-armv8Linux4.4gcc5.4.0.rtipkg
```
3. Generate an archive which contains the files required by `rmw_connextdds` (and other packages) to build:
```sh
tar czf rti_connext_dds-6.0.1-pro-armv8Linux4.4gcc5.4.0-rmw.tar.gz \
        rti_connext_dds-6.0.1/include \
        rti_connext_dds-6.0.1/lib/armv8Linux4.4gcc5.4.0 \
        rti_connext_dds-6.0.1/resource/cmake \
        rti_connext_dds-6.0.1/resource/scripts \
        rti_connext_dds-6.0.1/rti_versions.xml
```
4. Copy the generated archive to your build host, extract it, and use it to build `rmw_connextdds`:
```sh
# Copy archive and extract it
scp rti_connext_dds-6.0.1-pro-armv8Linux4.4gcc5.4.0-rmw.tar.gz my_build_host:~/
ssh my_build_host
tar xzf rti_connext_dds-6.0.1-pro-armv8Linux4.4gcc5.4.0-rmw.tar.gz

# Load Connext and ROS (e.g. foxy)
source /opt/ros/foxy/setup.bash
source rti_connext_dds-6.0.1/resource/scripts/rtisetenv_armv8Linux4.4gcc5.4.0.bash

# Build the RMW (e.g. already cloned in a workspace dir)
# You might need to specify CONNEXTDDS_ARCH and RTICONNEXTDDS_VERSION_OK
# to overcome some existing limitations in FindRTIConnextDDS.cmake.
# Also, you may receive warnings from CMake, but they should all be benign.
cd ws-rmw/
colcon build --cmake-args -DCONNEXTDDS_ARCH=armv8Linux4.4gcc5.4.0 -DRTICONNEXTDDS_VERSION_OK=true
```

As you noticed, we don't yet have an official manual for `rmw_connextdds`, but this is certainly something that will be useful to have in an article in our Knowledge Base.

Speaking of that, you might want to take a look at these two articles which try to address the lack of "native build" support for ARM platforms: one article outlines a [similar strategy (including a helper script) to generate an installation for Raspberry Pi from another host](https://community.rti.com/kb/how-do-i-create-connext-dds-application-rti-code-generator-and-build-it-my-arm-embedded-target), the other provides an [example of cross-compilation for Jetson Nano](https://community.rti.com/kb/using-rti-connext-dds-nvidia-jetson-nano) (which maybe could be followed for ROS2 using something like [ros-tooling/cross_compile](https://github.com/ros-tooling/cross_compile), but I have never tried -- also the KB only provides `makefile` instructions for Connext Pro, so an appropriate CMake toolchain file will need to be generated).

I hope this information is useful. Let me know if you have any more questions! I will wait for your confirmation before closing the issue.

asorbini, 2021-03-16:
I'm going to do a test of my proposed workflow, since I haven't actually tested it yet. I might have missed `rti_versions.xml` in my instructions.

spiderkeys, 2021-03-16:
We had to solve the same problem a month or so ago, so for whom it might be useful, here is a quick and dirty script that quickly creates host-lite release for a 'target' platform from a host installation and achieves what Andreas describes:

rtidist.sh
```
#!/usr/bin/env bash
set -e

function abs_path {
  (cd "$(dirname '$1')" &>/dev/null && printf "%s/%s" "$PWD" "${1##*/}")
}

# Handle options
for i in "$@"
do
case $i in
    -i=*|--input=*)
    CONNEXTDDS_DIR="$(readlink -f ${i#*=})"
    NDDSHOME=${CONNEXTDDS_DIR}
    CONNEXT_INSTALL_NAME=`basename ${CONNEXTDDS_DIR}`
    shift
    ;;

    -o=*|--output=*)
    OUTPUT_DIR="$(readlink -f ${i#*=})"
    shift
    ;;

    -a=*|--arch=*)
    CONNEXT_ARCH="${i#*=}"
    shift
    ;;

    *)
        # unknown option
    ;;
esac
done

if [ -z "${CONNEXTDDS_DIR}" ]; then
    echo "Input dir is not set! Use: --input=<dir>"
    exit 1
fi

if [ -z "${OUTPUT_DIR}" ]; then
    echo "Output dir is not set! Use: --output=<dir>"
    exit 1
fi

if [ -z "${CONNEXT_ARCH}" ]; then
    echo "Connext architecture is not set! Use: --arch=<arch>"
    exit 1
fi

####################

# First create the output directory
mkdir -p "${OUTPUT_DIR}"

BASE_DIRECTORY="${OUTPUT_DIR}/${CONNEXT_INSTALL_NAME}-${CONNEXT_ARCH}"
echo "Writing files to ${BASE_DIRECTORY}"

# Create basic required directory structure
mkdir -p "${BASE_DIRECTORY}"
mkdir -p "${BASE_DIRECTORY}/bin"
mkdir -p "${BASE_DIRECTORY}/lib"
mkdir -p "${BASE_DIRECTORY}/resource"

# Copy as-is folders
cp -r "${CONNEXTDDS_DIR}/include"               "${BASE_DIRECTORY}/"

# Copy top-level files
cp "${CONNEXTDDS_DIR}/rti_versions.xml"         "${BASE_DIRECTORY}/"

# Copy 'bin' files
cp "${CONNEXTDDS_DIR}/bin/rtiddsgen"            "${BASE_DIRECTORY}/bin/"
cp "${CONNEXTDDS_DIR}/bin/rtiddsgen_server"     "${BASE_DIRECTORY}/bin/"

# Copy 'lib' files
cp -r "${CONNEXTDDS_DIR}/lib/java"              "${BASE_DIRECTORY}/lib/"
cp -r "${CONNEXTDDS_DIR}/lib/${CONNEXT_ARCH}"   "${BASE_DIRECTORY}/lib/"

# Create 'resource' directories
mkdir -p "${BASE_DIRECTORY}/resource/app/app_support"
mkdir -p "${BASE_DIRECTORY}/resource/app/lib"
mkdir -p "${BASE_DIRECTORY}/resource/scripts"

# Copy 'resource' files
cp -r "${CONNEXTDDS_DIR}/resource/app/app_support/rtiddsgen"            "${BASE_DIRECTORY}/resource/app/app_support/"
cp -r "${CONNEXTDDS_DIR}/resource/app/lib/java"                         "${BASE_DIRECTORY}/resource/app/lib/"
cp -r "${CONNEXTDDS_DIR}/resource/cmake"                                "${BASE_DIRECTORY}/resource/"
cp -r "${CONNEXTDDS_DIR}/resource/idl"                                  "${BASE_DIRECTORY}/resource/"
cp -r "${CONNEXTDDS_DIR}/resource/schema"                               "${BASE_DIRECTORY}/resource/"
cp "${CONNEXTDDS_DIR}/resource/scripts/rticommon_config.sh"             "${BASE_DIRECTORY}/resource/scripts/"
cp "${CONNEXTDDS_DIR}/resource/scripts/rticommon.sh"                    "${BASE_DIRECTORY}/resource/scripts/"
cp "${CONNEXTDDS_DIR}/resource/scripts/rtisetenv_${CONNEXT_ARCH}."*     "${BASE_DIRECTORY}/resource/scripts/"
cp -r "${CONNEXTDDS_DIR}/resource/xml"                                  "${BASE_DIRECTORY}/resource/"
```
Usage:
```sh
 ./rtidist.sh -i=~/rti_connext_dds-6.0.1 -o=${PWD} -a=armv8Linux4.4gcc5.4.0
```
This will create a folder called `rti_connext_dds-6.0.1-armv8Linux4.4gcc5.4.0` with all of the files required to both be used by rmw_connextdds, as well as have everythng you need to use rtiddsgen on ARM64 as well (we use both on the Jetson).

asorbini, 2021-03-16:
Thank you for the contribution @spiderkeys, your script is definitely more high-level than my instructions and it could be useful to others!

Just FYI, if you are creating this version only to support compilation of `rmw_connextdds` then you could restrict the list of copied files to the ones I suggested in my previous comment:

- `${CONNEXTDDS_DIR}/include`
- `${CONNEXTDDS_DIR}/lib/${CONNEXTDDS_ARCH}`
- `${CONNEXTDDS_DIR}/resource/cmake`
- `${CONNEXTDDS_DIR}/resource/scripts`
- `${CONNEXTDDS_DIR}/rti_versions.xml`

This excludes `rtiddsgen` since it is no longer needed by `rmw_connextdds` (because it doesn't rely on `rosidl_typesupport_connext` like `rmw_connext_cpp`).

Your script is great in order to have installation that can build "native" Connext applications, which typically require code generation with `rtiddsgen` (unless they rely exclusively on the DynamicData API).

asorbini, 2021-03-16:
I'm closing the issue since I think we have a "solution" for now. Feel free to continue discussing the topic as needed.

closed 2021-03-16 by asorbini

Vermounthlly, 2022-08-11:
hi @asorbini  , I am trying to use RTI DDS with ROS2 on Raspberry Pi 4b.
I Downloaded "rti_connext_dds-6.1.1-lm-target-armv8Linux4gcc7.3.0.rtipkg" and “rti_connext_dds-6.1.1-lm-x64Linux4gcc7.3.0.run”,and I installed the armv8Linux package on a Linux host machine(PC) using the RTI Launcher as described in the [RTI document](https://community.rti.com/static/documentation/connext-dds/6.1.0/doc/manuals/connext_dds_professional/installation_guide/index.htm#installation_guide/Installing.htm%3FTocPath%3D_____3).

I use your method but  I got following errors:

    -- Detected CONNEXTDDS_DIR = '/opt/rti_connext_dds-6.1.1'
    -- CONNEXTDDS_ARCH not provided, trying to guess it...
    -- Guessed CONNEXTDDS_ARCH ('5gcc9.4.0') not available.
    -- Pick first from /opt/rti_connext_dds-6.1.1/lib/[armv8Linux4gcc7.3.0;java;x64Linux4gcc7.3.0]
    -- unsupported CMAKE_HOST_SYSTEM_NAME (Linux) or CMAKE_HOST_SYSTEM_PROCESSOR (aarch64). Using architecture armv8Linux4gcc7.3.0 anyway.
    -- Selected CONNEXTDDS_ARCH: armv8Linux4gcc7.3.0
    -- RTI Connext DDS installation directory: /opt/rti_connext_dds-6.1.1
    -- RTI Connext DDS architecture: armv8Linux4gcc7.3.0
      CMake Error at /opt/rti_connext_dds-6.1.1/resource/cmake/FindRTIConnextDDS.cmake:1102 (message):
      Linux-5.4.0-1066-raspi is not supported as host architecture
      Call Stack (most recent call first):
      cmake/rti_build_helper.cmake:572 (find_package)
      CMakeLists.txt:25 (rti_find_connextpro)

    -- Configuring incomplete, errors occurred!
    See also "/home/lly/ros2_connextdds/build/rti_connext_dds_cmake_module/CMakeFiles/CMakeOutput.log".
    Failed   <<< rti_connext_dds_cmake_module [3.10s, exited with code 1]
    lly@lly:~/ros2_connextdds$ 

Is there any suggestion? Thank you!

sumitpaulde, 2023-01-17:
Hi @asorbini and @Vermounthlly  Do you have any update regarding this issue? I am struggling with the same error on Xavier.

gabriel-st-angel-aerospace, 2025-04-21:
For anyone interested, I have been installing the binaries and libraries for arm platforms using the apt repository.  It's confusing that RTI doesn't provide a .run package for arm, but has arm binaries in apt.  This is the easiest solution for me at the moment until RTI provides a arm .run.

[RTI Apt Install instructions ](https://community.rti.com/static/documentation/developers/get-started/apt-install.html)