ros2/rmw_connextdds #24: New release into Rolling

clalancette, 2021-04-06:
@asorbini Could you please do a new release of rmw_connextdds into Rolling?  That should help us get the packages building with all of the recent changes for unique network flows and the like.  Thanks.

clalancette, 2021-04-06:
I should also add: if you don't have time for it today, just let me know.  I can do the release as well.

asorbini, 2021-04-06:
@clalancette I was hoping to merge #22 before a new release, and bump the version number to 0.5.0, but from test results and @ivanpauno's review, I see I still have some work on that, and probably won't be able to merge it before tomorrow at the earliest, to allow for any additional review (assuming I'll be able to get the code in final shape today).

If you can do the release today I would appreciate it. Maybe bump the version to 0.4.1? (or 0.5, and we'll go to 0.6 after #22)

Thanks!

clalancette, 2021-04-06:
Sounds good, I'll go ahead and do it.  Numbers are cheap, we can always bump again :).  Thanks for the reply.

asorbini, 2021-04-14:
@clalancette I'm going to let CI run overnight to validate the recent changes (#22, #28). If everything is as expected, I'll go ahead and create `0.6.0` tomorrow.

asorbini, 2021-04-14:
CI was green/as expected, going ahead with 0.6.0 release now.

For reference:
- nightly_linux_release: [image]
- nightly_osx_release: [image]
- nightly_win_rel: [image]
- Rci__nightly-connext_ubuntu_focal_amd64: [image]

asorbini, 2021-04-14:
Tagged and bloomed `0.6.0` (ros/rosdistro#29138)

closed 2021-04-14 by asorbini