ros2/rmw_connextdds #27: Cannot run demo node using RMW Connext DDS 6.0.1

eranroll, 2021-04-12:
Hi,

I have downloaded RTI Connext DDS 6.0.1 from the RTI downloads website, and also compiled the src of rmw_connextdds using colcon: `colcon build --symlink-install --packages-up-to rmw_connextdds`
When I try to use the RTI Connext RMW, it fails with the following error:
[image]

This is what I am trying to run: `RMW_IMPLEMENTATION=rmw_connextdds ros2 run demo_nodes_cpp talker`

I am running ROS Eloquent, and I have pre-sourced all the required scripts.

Any ideas on what I am doing wrong? Is there a package missing?

asorbini, 2021-04-12:
Hi @eranroll,

thank you for your report.

I have identified and resolved a couple of issues in the `eloquent` branch which had unfortunately escaped my review during the a recent merge to the branch (#18).

Please update to the latest commit ([a64051f](https://github.com/ros2/rmw_connextdds/commit/a64051f340089793ebbbcd0cb7dafe2e1ea1e4fb)) and give it another try. I recommend doing a clean build (i.e. please run `rm -r build/ install/` in the workspace when you are building `rmw_connextdds` and start from a "clean environment" where you have only loaded ROS 2 and RTI Connext DDS).

Let me know if that fixes the issues, or if you still have any problems.

eranroll, 2021-04-13:
It fixed it. Thank you very much!

closed 2021-04-13 by eranroll