ros2/rmw_connextdds #46: New minor release into Rolling/Galactic

clalancette, 2021-04-26:
@asorbini I was going to do a new minor release into both rolling and galactic, 0.6.1.  This will get one more fix into Galactic that is pending, and it is just easier at this point to keep Rolling >= to Galactic.  Any objections?

asorbini, 2021-04-26:
Sounds good, thank you for the heads up.

I'm assuming the fix you are referring to is #43?

clalancette, 2021-04-26:
> I'm assuming the fix you are referring to is #43?

Ah, no, it looks like the fix that is already committed is #33; that's the one I wanted to release to both Galactic and Rolling.  Still OK?

asorbini, 2021-04-26:
Oh right, forgot about that. Still all good 👍🏼

clalancette, 2021-04-26:
https://github.com/ros/rosdistro/pull/29306 and https://github.com/ros/rosdistro/pull/29307 are open.  They aren't merged yet, but I'll do a pass a little later on (or tomorrow morning) and merge them.  So I'll close this out, thanks for the feedback!

closed 2021-04-26 by clalancette