ros2/rmw_connextdds #59: New release

ivanpauno, 2021-06-02:
I need a new release after having merged https://github.com/ros2/rmw_connextdds/pull/20.
If not, the rolling PR checker isn't going to pass in the rmw implementation repo https://github.com/ros2/rmw_implementation/pull/188.

@asorbini I can do it, but I wanted to double if that's ok or if you prefer to do it.

asorbini, 2021-06-02:
@ivanpauno thank you for checking! If you can, go ahead with the release. Keep in mind that for Rolling, we should bump the minor release number, as pointed out by @clalancette in [this comment](https://github.com/ros2/rmw_connextdds/issues/58#issue-897397176).

ivanpauno, 2021-06-04:
Done https://github.com/ros/rosdistro/pull/29771.

closed 2021-06-04 by ivanpauno