ros2/rmw_connextdds #60: ROS won't find rmw_connextdds when not installed at ~/

AlonSheaderAtForetellix, 2021-06-15:
**System Info**

- _OS_
  - Ubuntu 20.04
- *ROS version and installation type*
  - foxy/source
- *RTI Connext DDS version and installation type*
  - Pro 6.0.1
- *RMW version or commit hash*
  - 46b23ca3ca718ab4c624898b1e543b0b3e57000d

**Bug Description**

When following the quick installation guide, 
the demo will work for installing in ~/ros2_connextdds/src/ros2
But not in any different directory.

**Expected Behavior**

The demo should work.

**How to Reproduce**

Follow the quick guide installation but install the ws in different dir.

```

```

**Workarounds**

[Ways that you have discovered to overcome the issue, if any]

**Additional context**

[Any additional relevant information about the problem]

asorbini, 2021-06-15:
@AlonSheaderAtForetellix can you please provide more information on your issue?

I'm not sure I understand it, and I'm unable to reproduce it. I just tried a quick test, and the following worked for me without problems:

```sh
# Create a custom ROS workspace and enter it
mkdir -p /tmp/foo/bar
cd /tmp/foo/bar
# Clone the RMW package anywhere inside the workspace
git clone -b foxy https://github.com/ros2/rmw_connextdds
# Load ROS 2 Foxy
source /opt/ros/foxy/setup.bash
# Load Connext DDS 6.0.1 (e.g. x64Linux4gcc7.3.0)
source /path/to/rti_connext_dds-6.0.1/resource/scripts/rtisetenv_x64Linux4gcc7.3.0.bash
# Build RMW with colcon
colcon build --symlink-install
# Load workspace
source install/setup.bash
# Spawn a hello world publisher
RMW_IMPLEMENTATION=rmw_connextdds ros2 run demo_nodes_cpp talker
```

In general, you can build and load the RMW like any other ROS 2 package, i.e. by placing it inside a "workspace" directory and running `colcon build`, then loading the workspace's artifacts with the `setup.bash` script.

AlonSheaderAtForetellix, 2021-06-15:
@asorbini 

Thanks you for the quick reply.
I am not able to reproduce my issue.
Things are working as expected, I guess I had to get some of your MOJO for run it properly :)

Closing.

closed 2021-06-15 by AlonSheaderAtForetellix