ros2/rmw_connextdds #63: should return if CONNEXTDDS_ARCH is not set after guessing

asobhy-qnx, 2021-08-20:
https://github.com/ros2/rmw_connextdds/blob/46b23ca3ca718ab4c624898b1e543b0b3e57000d/rti_connext_dds_cmake_module/cmake/rti_build_helper.cmake#L562

should be:
```
        if(NOT CONNEXTDDS_ARCH)
          rti_guess_connextdds_arch()
          if(NOT CONNEXTDDS_ARCH)
            return()
          endif()
        endif()
```

asorbini, 2021-09-01:
Hi @asobhy-qnx,

Thank you for your suggestion! The change makes sense to me, since at the moment the build will "fall through" without a `CONNEXTDDS_ARCH` value, and fail more subtly at a later point.

I'll try to update the build files soon, but feel free to open a PR on the repository if you want/have the bandwidth. Community contributions are always welcome 😄

fgallegosalido, 2026-09-10:
@asobhy-qnx we are planning on removing support for Connext earlier than 6.X, and this guessing logic will not be needed anymore. I am closing this issue.

closed 2026-09-10 by fgallegosalido