ros2/rmw_connextdds #64: Message compatibility to native DDS

millerluki, 2021-08-31:
**System Info**

- _OS_
  - Ubuntu 20.04
- *ROS version and installation type*
  - galactic/binary - apt
- *RTI Connext DDS version and installation type*
  - rti-connext-dds-5.3.1/apt
- *RMW version or commit hash*
  - ros-galactic-rmw-connextdds: 0.6.2-1focal.20210630.224436

**Bug Description**

Generated *.idl files from colcon & rosidl command line tool do not match with the observed messages in RTI Admin Console and do not match our native DDS participants build on the ROS generated *.idl files.

In Foxy I have the middleware specific *.idl files in the `/opt/ros/foxy/share/PKG/msg/dds_connext` directory of ROS2 which can be used with the RTI toolchain. 

In Galactic there are no middleware specific *.idl files. The ros publishers produce unexpected message definitions observed in the RTI tools when compared to the generic *.idl files. Therefore, the generic *.idl files cannot be used in our native DDS counterpart application.

**Expected Behavior**

Generated *.idl from colcon & rosidl command line tool can be used in RTI Code Generator (rtiddsgen) and result in data type match.

**How to Reproduce**

* source ROS2 Galactic & RTI Environments
* `export RMW_IMPLEMENTATION=rmw_connextdds`
* `ros2 topic pub /test std_msgs/msg/Float64 'data: 1.234'`
* open RTI Admin Console and analyse message definition sent on topic _/test_

**Workarounds**

Usage of ROS2 Foxy message types for RTI Connext DDS middleware in native DDS application. In Galactic we set the legacy flag for rmw_connextdds
`export RMW_CONNEXT_LEGACY_RMW_COMPATIBILITY_MODE=1`

**Additional context**

[Any additional relevant information about the problem]

asorbini, 2021-09-01:
Hi @millerluki,

Thank you for your report. First of all let me try to shed some light on this apparent naming inconsistency, which stems from some design and implementation decisions in ROS2 that are not specific to the Connext RMW.

In particular, it looks like during the implementation of the other Tier 1 RMW implementations and the old implementation for Connext (`rmw_connext_cpp`), the choice was made to add the `dds_` namespace and an underscore suffix to the name of the ROS2 types registered with the DDS DomainParticipant created by the RMW. I'm not sure of the motivations behind this choice, but when it came to creating `rmw_connextdds`, we were forced to adopt this same naming convention in order to maintain interoperability with other RMW implementations.

This is because, by default, the DDS specification allows for a simpler matching logic to be employed by an implementation when trying to pair two endpoints. This algorithm ignores the actual type description and it instead only considers the name of the topic, and the name of the data type, which must match exactly as strings. It is assumed by both parties that the two applications are actually using the same type definition and that they have not in fact registered different types with the same name.

The "Extensible types for DDS" specification introduced an improved matching algorithm which instead relies on a standardized type description (called TypeObject) that applications can exchange over the wire (as part of DDS' discovery phase). This description can be used by the DDS implementations on both sides to determine if a remote endpoint will be able to produce/consume data samples compatible with a local endpoint. If this information is not available (e.g. if the remote application's DDS implementation does not support propagation of standard TypeObjects), then the DDS implementation will fall back to the "legacy" matching algorithm.

Connext supports propagation of TypeObjects via discovery and the TypeObject-based matching algorithm, so the difference of type names should not prevent matching. I'm not sure of which other DDS implementations also support these same features, but if an implementation only supports legacy matching, it would not be able to match an application that uses one the definition from the ROS2-generated IDL files, because, as of now, these files will not include the `dds_` namespace nor the underscore suffix, as you have noticed.

As you have also noted, this inconsistency could be overcome in the past by making use of the other set of IDL files which used to be generated by `rmw_connext_cpp`. These type definitions did include the type name mangling, but they are no longer needed nor generated, since `rmw_connextdds` uses a different serialization/deserialization strategy which does not require any code generation with `rtiddsgen`.

Nevertheless, **those alternative type definitions are not required to successfully establish communication between a ROS2 application running on `rmw_connextdds` and a "native" DDS application, running on Connext 5.3.1 or later**.

I did a few tests using the command that you provided and a (minimally modified) "hello world" application generated with `rtiddsgen` from the default IDL generated by ROS2 (i.e. the one without the name mangling):

```sh
rtiddsgen -language C++11 -example ${CONNEXTDDS_ARCH} /opt/ros/galactic/share/std_msgs/msg/Float64.idl
```

The generated subscriber (`Float64_subscriber.cxx`) was manually modified on line 60 to use the appropriate topic name (`"rt/test"`).

What I noticed is that communication works as expected if both applications are using the same Connext version, but that no communication occurred if they use different Connext versions, which might match your experience. This is because Connext 6.0.1 introduced "TypeObject compression", which optimizes the discovery process but is incompatible with earlier versions, and enabled by default. For this reason, a Connext 6.0.1 application will not be able to send type information to a Connext 5.3.1 unless the DomainParticipant's QoS is configured to disable this compression. Once this change is applied on the 6.0.1 side, type descriptions can be shared between the applications and communication is successfully established even with incompatible type names.

I have attached the XML configuration file which I used to disable type compression in 6.0.1 ([USER_QOS_PROFILES.xml](https://github.com/ros2/rmw_connextdds/files/7087344/USER_QOS_PROFILES.xml.txt), renamed to `.txt` since GitHub doesn't allow XML files to be attached).

## Test: 5.3.1 vs 5.3.1

1. Install ROS2 Galactic and `rmw_connextdds` (and RTI Connext DDS 5.3.1 as a dependency).

   ```sh
   sudo apt install ros2-galactic-desktop ros2-galactic-rmw-connextdds
   ```

2. Load ROS2 and Connext:

   ```sh
   source /opt/ros/galactic/setup.bash
   source /opt/rti.com/rti_connext_dds-5.3.1/resource/scripts/rtisetenv_x64Linux3gcc5.4.0.bash
   ```

3. Generate a "hello world" application with `rtiddsgen`.

   ```sh
   mkdir hello-float64
   rtiddsgen -language C++11 -example ${CONNEXTDDS_ARCH} -d hello-float64 /opt/ros/galactic/share/std_msgs/msg/Float64.idl
   ```

4. Modify `hello-float64/Float64_subscriber.cxx` on line 60, and change the topic name from `"Example std_msgs_msg_Float64"` to `"rt/test"`. 

5. Build the "hello world" applications.

   ```sh
   cd hello-float64
   make -f makefile_Float64_${CONNEXTDDS_ARCH}
   ```

[... cut ...]
I apologize for the "wall of text" but I hope to have provided some useful information. To summarize, I'm unable to exactly reproduce the behavior that you described. While it is unfortunate that ROS2 currently contains this naming inconsistency (which could be easily resolved by either getting rid of the unnecessary name mangling at the RMW layer, or by "propagating it" to the automatically generated IDL files), applications running on `rmw_connextdds` and/or RTI Connext DDS 5.3.1+ have several options to overcome this limitation and allow matching even with mismatching type names/descriptions.

Please let me know if I missed something and/or if you have any further questions/issues.

millerluki, 2021-09-02:
Thank you for your quick and detailed answer. I will have a try to the various options over the next few days and will give you feedback.