ros2/rmw_connextdds #70: rclcpp::SensorDataQoS not working for sensor_msgs::msg::Images

xander-m2k, 2022-01-31:
**System Info**

- _OS_
  - Ubuntu 20.04
- *ROS version and installation type*
  - ROS Foxy binary install
- *RTI Connext DDS version and installation type*
  - Newest apt packages Foxy as of today

**Bug Description**

When publishing sensor_msgs::msg::Image with images larger than h*w (tested with height 1944 width 2592), and using the standard QoS for sensors, messages cannot be read. Messages are published (checked with wireshark).

**Expected Behavior**

Messages should be readable.

**How to Reproduce**

Publish an image of a similar size I described (with sensor_msgs::msg::Image and rclcpp::SensorDataQoS), and read on that topic (ros2 topic echo /<image_topic>. Cannot share code, working on code under NDA, I'm sorry.

```

```

**Workarounds**

Just used the standard '10' as QoS with the `create_publisher()` function in rclcpp

asorbini, 2022-02-01:
Hi @xander-m2k,

thank you for reporting this issue. I will try to reproduce it soon, but on the top of my head, I suspect the issue might lie in the use of "best effort" Reliability QoS ([used by the "sensor data" profile](https://github.com/ros2/rmw/blob/de38e79ca8906c8a7757557acdef57f8474351ff/rmw/include/rmw/qos_profiles.h#L25)) when trying to send "large data" (i.e. messages which need DDS-level fragmentation like the image you are trying to publish).

I'll quote from [this online documentation page](https://community.rti.com/kb/what-do-i-need-send-large-data-successfully):

> If you use Best Effort, the application is not going to try to recover any lost fragments. Since in Best Effort communication the lost fragment isn’t resent, the DataReader will discard the entire sample. Depending on its size, the sample could have a lot of fragments, meaning it is more likely to lose a fragment (and therefore, the entire sample). By using Reliable Reliability, if a fragment is lost Connext DDS will try to recover it.

I would suggest switching the publisher to be "reliable" and see if that improves things.

> **Workarounds**
> 
> Just used the standard '10' as QoS with the `create_publisher()` function in rclcpp

I'm not quite sure how modifying the history depth would affect this issue. I suspect that by "standard '10'" you mean you are using the alternative signature of `create_publisher()` which only takes the history depth value and applies it on top of the standard QoS profile. It would make sense that you see communication happen in this case, because [the default QoS is reliable](https://github.com/ros2/rmw/blob/de38e79ca8906c8a7757557acdef57f8474351ff/rmw/include/rmw/qos_profiles.h#L51).

fgallegosalido, 2026-09-11:
Hi @xander-m2k. As this is quite old, did the suggestions help you fix the issue? As @asorbini correctly mentioned, large data and Best Effort don't really work well together, as losing a single fragment means losing the whole sample.

xander-m2k, 2026-09-11:
Hi @fgallegosalido, this is quite a while ago indeed, worked for another company back then. I ended up creating another issue here:https://github.com/ros2/ros2/issues/1242
and a PR here: https://github.com/eProsima/Fast-DDS-docs/pull/341
Maybe that will help, good luck!

fgallegosalido, 2026-09-15:
In that case, I will close this issue, as this seems an issue more related to the use of large data rather than the RMW itself. Thank you for your collaboration!

closed 2026-09-15 by fgallegosalido