ros2/rmw_connextdds #75: Exposing the configuration to force local traffic on loopback

MartijnVogelaar, 2022-03-08:
**Exposing the configuration to force local traffic on loopback**

In the previous version of the RMW [rmw_connext](https://github.com/ros2/rmw_connext) the local traffic was, by default, forced to be sent over loopback as can be seen [here.](https://github.com/ros2/rmw_connext/blob/1386fd6ca29b10a3b7c2214af31a5b68c0d0e20b/rmw_connext_shared_cpp/src/node.cpp#L143-L151)

This allowed me to capture the traffic between nodes running on the same system. It currently seems to be the case that shared memory is used which does not allowed me to capture the traffic between the nodes.

Looking at the current implementation I saw the following piece of code
https://github.com/ros2/rmw_connextdds/blob/92234d16fa257048c9db5bc0d92241fe72069fcb/rmw_connextdds_common/include/rmw_connextdds/static_config.hpp#L295-L303

The description of the define seems to be what I need but it doesn't seem to be implemented yet. Is this something that is going to be implemented and possibly be exposed as (runtime) configuration anytime soon, if it is going to be implemented at all?

It might be the case that I am missing something and that it is possible to capture traffic between two nodes on the same system, if this is the case another solution is fine as well.

asorbini, 2022-03-16:
Hi @MartijnVogelaar,

That macro is a left over from some code that has since been delete. As you correctly guessed, the code had been initially ported from `rmw_connext_cpp` to support the duplication of data over loopback. As I understand it, this configuration was used by `rmw_connext_cpp` to overcome some interoperability issues with other RMWs, which have since been resolved.

Additionally, the code was removed because we observed that the decision to always duplicate all traffic on the loopback interface had significant impact on performance.

Since the use case didn't seem compelling for most users, we decided to avoid supporting that functionality at the RMW level. 

Users can achieve a somewhat similar (but more efficient) behavior by disabling the shared-memory transport and forcing all local communications to occur over UDPv4/loopback.

Connext can be configured via XML (loaded in [multiple ways](https://community.rti.com/static/documentation/connext-dds/6.0.1/doc/manuals/connext_dds/html_files/RTI_ConnextDDS_CoreLibraries_UsersManual/index.htm#UsersManual/How_to_Load_XML_Specified_QoS_Settings.htm#xmlconfiguration_1275484337_316960%3FTocPath%3DPart%25203%253A%2520Advanced%2520Concepts%7C19.%2520Configuring%2520QoS%2520with%2520XML%7C19.5%2520How%2520to%2520Load%2520XML-Specified%2520QoS%2520Settings%7C_____0)). For example, in this case, you could use an XML file with the following content:

```XML
<dds>
  <qos_library name="ExampleLibrary">
    <qos_profile name="ExampleProfile" is_default_qos="true">
      <participant_qos>
        <transport_builtin>
          <mask>UDPv4</mask>
        </transport_builtin>
      </participant_qos>
    </qos_profile>
  </qos_library>
</dds>
```

If you still want to use the shared-memory transport and replicate the exact same behavior as `rmw_connext_cpp`, you can remove the `<transport_builtin>` element, and add the following to `<participant_qos>`:

```xml
<property>
  <value>
    <element>
      <name>dds.transport.UDPv4.builtin.ignore_loopback_interface</name>
      <value>0</value>
    </element>
  </value>
</property>
```

You can then load it into your ROS 2 application by either placing in the "current working directory" and naming it `USER_QOS_PROFILES.xml`, or by placing it somewhere else and/or naming it something else, and then specifying it's path through environment variable `NDDS_QOS_PROFILES`.

I hope this helps you achieve your goal. Please let me know if you have any questions/problems.

MartijnVogelaar, 2022-03-17:
Hi @asorbini 

Thanks for the reply! I already found another workaround but thanks nevertheless!

Consider the issue solved!

closed 2022-03-17 by MartijnVogelaar