ros2/rmw_connextdds #91: How to build  rmw_connextdds on Raspberry Pi 4b ?

Vermounthlly, 2022-08-09:
Hi, I am trying to use RTI DDS with ROS2.
System Info

    OS
    Ubuntu20.04  
    ROS version and installation type
    foxy
    RTI Connext DDS version and installation type
    6.1.1

In desktop environment, I succeed with installing RTI DDS 6.1.1 with rmw_connextdds repository.

Now I am trying to install RTI DDS on Raspberry Pi 4b (armv8 architecture).

I Downloaded "rti_connext_dds-6.1.1-lm-target-armv8Linux4gcc7.3.0.rtipkg" and “rti_connext_dds-6.1.1-lm-x64Linux4gcc7.3.0.run”,and 
I installed the armv8Linux package on a Linux host machine(PC) using the RTI Launcher as described in the [RTI document](https://community.rti.com/static/documentation/connext-dds/6.1.0/doc/manuals/connext_dds_professional/installation_guide/index.htm#installation_guide/Installing.htm%3FTocPath%3D_____3).

Then I copied the whole installation(/opt/rti_connext_dds-6.1.1/) to the Raspberry Pi 4b and to build rmw_connextdds  refer to #10 ,
but  I got following errors:

    lly@lly:~/ros2_connextdds$ colcon build
    Starting >>> rti_connext_dds_cmake_module
    -- The C compiler identification is GNU 9.4.0
    -- The CXX compiler identification is GNU 9.4.0
    -- Check for working C compiler: /usr/bin/cc
    -- Check for working C compiler: /usr/bin/cc -- works
    -- Detecting C compiler ABI info
    -- Detecting C compiler ABI info - done
    -- Detecting C compile features
    -- Detecting C compile features - done
    -- Check for working CXX compiler: /usr/bin/c++
    -- Check for working CXX compiler: /usr/bin/c++ -- works
    -- Detecting CXX compiler ABI info
    -- Detecting CXX compiler ABI info - done
    -- Detecting CXX compile features
    -- Detecting CXX compile features - done
    -- Found ament_cmake: 0.9.9 (/opt/ros/foxy/share/ament_cmake/cmake)
    -- Found PythonInterp: /usr/bin/python3 (found suitable version "3.8.10", minimum required is "3") 
    -- Using PYTHON_EXECUTABLE: /usr/bin/python3
    -- Detected CONNEXTDDS_DIR = '/opt/rti_connext_dds-6.1.1'
    -- CONNEXTDDS_ARCH not provided, trying to guess it...
    -- Guessed CONNEXTDDS_ARCH ('5gcc9.4.0') not available.
    -- Pick first from /opt/rti_connext_dds-6.1.1/lib/[armv8Linux4gcc7.3.0;java;x64Linux4gcc7.3.0]
    -- unsupported CMAKE_HOST_SYSTEM_NAME (Linux) or CMAKE_HOST_SYSTEM_PROCESSOR (aarch64). Using architecture armv8Linux4gcc7.3.0 anyway.
    -- Selected CONNEXTDDS_ARCH: armv8Linux4gcc7.3.0
    -- RTI Connext DDS installation directory: /opt/rti_connext_dds-6.1.1
    -- RTI Connext DDS architecture: armv8Linux4gcc7.3.0
      CMake Error at /opt/rti_connext_dds-6.1.1/resource/cmake/FindRTIConnextDDS.cmake:1102 (message):
      Linux-5.4.0-1066-raspi is not supported as host architecture
      Call Stack (most recent call first):
      cmake/rti_build_helper.cmake:572 (find_package)
      CMakeLists.txt:25 (rti_find_connextpro)

    -- Configuring incomplete, errors occurred!
    See also "/home/lly/ros2_connextdds/build/rti_connext_dds_cmake_module/CMakeFiles/CMakeOutput.log".
    Failed   <<< rti_connext_dds_cmake_module [3.10s, exited with code 1]
    lly@lly:~/ros2_connextdds$ 

Is there any suggestion???
Thank you !!!!!

sumitpaulde, 2023-01-16:
Hi there, do you have any update regarding this issue? I am facing the same error.

asorbini, 2023-01-23:
The issue stems from script `resource/cmake/FindRTIConnextDDS.cmake` included with Connext 6.1.1, which unfortunately does not support armv8 "host" environments.

I tried replacing it with the latest version (available from the [rticommunity/rticonnextdds-cmake-utils repository](https://github.com/rticommunity/rticonnextdds-cmake-utils/blob/main/cmake/Modules/FindRTIConnextDDS.cmake)) and I was able to build `rmw_connextdds` successfully. I suggest downloading this version (e.g. [from the "raw" link](https://raw.githubusercontent.com/rticommunity/rticonnextdds-cmake-utils/main/cmake/Modules/FindRTIConnextDDS.cmake)) and replacing the one included by default.

Note that if you have more than one set of "target libraries" installed under `$CONNEXTDDS_DIR/lib`, you will have to specify which "architecture" to use  by passing `CONNEXTDDS_ARCH` to `cmake` (e.g. `colcon build --cmake-args -DCONNEXTDDS_ARCH=armv8Linux4gcc7.3.0`).

As I mentioned in #10, if you are using Connext only to support `rmw_connextdds`, you could also copy just the required subset of files from the x64 Linux installation in order to reduce the size of the installation on the Raspberry Pi.

You can achieve this with a command like:

```sh
tar czf rti_connext_dds-6.1.1-armv8Linux4gcc7.3.0-ros-runtime.tar.gz \
        rti_connext_dds-6.1.1/include \
        rti_connext_dds-6.1.1/lib/armv8Linux4gcc7.3.0 \
        rti_connext_dds-6.1.1/resource/cmake \
        rti_connext_dds-6.1.1/resource/scripts \
        rti_connext_dds-6.1.1/rti_versions.xml
```

Then copy `rti_connext_dds-6.1.1-armv8Linux4gcc7.3.0-ros-runtime.tar.gz` and extract it on the Raspberry Pi.

closed 2023-01-23 by asorbini