ros2/rmw_connextdds #94: Currently used middleware can't loan messages. Local allocator will be used.

ceccocats, 2022-10-24:
Is ros2 loaned message interface supported?

I have nodes that sends static messages with zero-copy using the loaned inteface like this:
https://github.com/ros2/demos/blob/rolling/demo_nodes_cpp/src/topics/talker_loaned_message.cpp

Is it possible to use it with RTI Connext DDS with the performace expected with a zero copy DDS?

ZhenshengLee, 2023-01-03:
@ceccocats there is a description in the official blog of RTI about zero copy of connextdds in ros2

> Incidentally, Connext DDS also supports a Zero Copy mode that can get the increased performance when sending large data inside a Process or over shared memory. Moreover, the Zero Copy approach used by Connext does not "bypass" the serialization and protocol and therefore does not cause the undesirable “coupling” side-effects. However, the Connext Zero Copy utilizes a specialized API that cannot be mapped to the ROS RMW API. You can explore the benefits of the Zero Copy feature as well as more extensive performance scenarios using rtiperftest (the RTI performance benchmarking tool), and the results of a [previous RTI blog](https://www.rti.com/blog/the-indy-autonomous-challenge-achieving-extreme-performance-with-ros-2).

the source is in https://www.rti.com/blog/latest-connext-dds-ros-2-performance-benchmarks

thanks.

asorbini, 2023-01-20:
`rmw_connextdds` does not currently support the "loan message" API exposed by ROS 2.

As already pointed out, RTI Connext DDS does support distribution of data using zero-copy semantics but only when the native DDS API is used.

You can find [more information about this feature in the Connext user manual](https://community.rti.com/static/documentation/connext-dds/7.0.0/doc/manuals/connext_dds_professional/users_manual/users_manual/SendingLDZeroCopyUsing.htm#34.1.5.1_Using_Zero_Copy_Transfer_Over_Shared_Memory).

Note that in order to use zero copy, you will need to create "native" DDS endpoints (i.e. DataReaders and DataWriters), and define your message in a format that can be processed by RTI's code generator, `rtiddsgen` (e.g. IDL, or XML).

I'm closing this issue but I opened #98 to track requests for this feature.

closed 2023-01-20 by asorbini