ros2/rmw_connextdds #100: Environment variable CONNEXTDDS_ARCH not detected by build script

asorbini, 2023-01-23:
**Bug Description**

The CMake build script does not correctly detect variable `CONNEXTDDS_ARCH` from the shell environment, and thus the build may fail if multiple target architectures have been installed inside `CONNEXTDDS_DIR` because the wrong architecture is automatically selected.

**Expected Behavior**

Variable `CONNEXTDDS_ARCH` is correctly picked up from the shell environment, and used by the build script to determine which Connext architecture to use.

**Workarounds**

Pass `CONNEXTDDS_ARCH` as a CMake argument instead of an environment variable:

```sh
colcon build ... --cmake-args -DCONNEXTDDS_ARCH=<architecture>
```

[The initial check](https://github.com/ros2/rmw_connextdds/blob/dcf6eddf21d24cde50f7b300d637301028622f03/rti_connext_dds_cmake_module/cmake/rti_build_helper.cmake#L408) inside helper function `rti_guess_connextdds_arch()` must be changed to something like:

```cmake
if(NOT "$ENV{CONNEXTDDS_ARCH}" STREQUAL "")
```

cwecht, 2023-05-03:
https://github.com/ros2/rmw_connextdds/pull/105 should fix this issue.

clalancette, 2023-06-13:
Closing now that #105 is merged.

closed 2023-06-13 by clalancette