ros2/rmw_connextdds #112: Connext 7.0.0

BeatScherrer, 2023-04-02:
**Feature Description**

Adjust package to suppor rti dds connext 7.0.0

**Implementation Considerations**

When compiling as is the following errors occur:

```
dds_api_ndds.cpp: In function ‘rmw_ret_t rmw_connextdds_return_samples(RMW_Connext_Subscriber*)’:
```
```
dds_api_ndds.cpp:818:7: error: invalid conversion from ‘DDS_Long’ {aka ‘int’} to ‘DDS_SampleInfoSeq*’ [-fpermissive]
```
```
dds_api_ndds.cpp:815:40: error: too many arguments to function ‘DDS_ReturnCode_t DDS_DataReader_return_loan_untypedI(DDS_DataReader*, void**, DDS_SampleInfoSeq*)’
```

mjcarroll, 2023-04-13:
Since this would be bumping a major version of a large dependency, this would likely need to be part of the roadmap for the next version of ROS 2 (J-turtle).

I'm going to backlog this for now and we can consider this with the RTI folks as a potential feature for the next release.

CC: @asorbini

asorbini, 2023-09-25:
Fixed by #121

closed 2023-09-25 by asorbini

BeatScherrer, 2023-09-25:
Awesome! thanks for your efforts @asorbini