ros2/rmw_connextdds #119: Support Connext key fields

superware, 2023-05-07:
Hello,

I have a working ROS2 foxy ecosystem running Connext and came to the point where I need Connext key fields support. I'm looking for the simplest way to modify ROS2 or rmw_connextdds and allow the underlaying Connext RMW to have some of the fields defined as keys.

Most of the related rmw implementation seem to be in [rmw_connext_dynamic_cpp/src/utility_templates.hpp](https://github.com/ros2/rmw_connext/blob/foxy/rmw_connext_dynamic_cpp/src/utility_templates.hpp) which in conjunction with [DDS_StructMember.is_key](https://community.rti.com/static/documentation/connext-dds/6.1.2/doc/api/connext_dds/api_c/structDDS__StructMember.html) seems technically possible but I don't have the necessary architectural knowledge to workaround.

Any help will be greatly appreciated.

clalancette, 2023-05-18:
There is no current plan to work on this.  This is actually a pretty complicated feature request, so if you were interested in pursuing it please start by opening a design document REP on https://github.com/ros-infrastructure/rep .

For now I'll put this on the backlog.

superware, 2023-05-18:
I'm not referring to ROS support for key fields, but rather to somehow add the is_key bit in Connext RMW typesupport. Which issues would rise in ROS if the RMW will internally handle keyed topics?

fgallegosalido, 2026-09-11:
Hi @superware , we already added support for keyed topics in Kilted (see https://github.com/ros2/rmw_connextdds/pull/178). The way you can benefit from this feature is by generating an IDL file from the original .msg file, and then modifying the generated IDL to include the `@key` annotation to the members you want to be marked as keys. You can follow this tutorial on how to work with them: https://docs.ros.org/en/rolling/ROS-Framework/interfaces/topics/Working-with-topics/Topic-Keys/Topic-Keys-Tutorial.html.

I'm closing this issue, but feel free to open a new one or reopen this one if you have any other doubts.

closed 2026-09-11 by fgallegosalido