ros2/rmw_connextdds #127: connext dds micro does not compile with rolling branch

cwecht, 2023-07-31:
**System Info**

- _OS_
  - Ubuntu 22.04
- *ROS version and installation type*
  - rolling/binary
- *RTI Connext DDS version and installation type*
  - Micro 3.0.3.28/rtipkg
- *RMW version or commit hash*
  - fff0668a7bc816a9075736a7c0ac08f124f394f9

**Bug Description**

rmw_connextdds_micro does not compile on rolling because of these errors:

[image]

It seems that connext dds micro just doesn't support DDS_UserDataQosPolicy as well as the `multicast_receive_addresses` option of `DDS_DiscoveryQosPolicy`.

**Expected Behavior**

rmw_connextdds_micro should compile.

**How to Reproduce**

That's my setup:

1. setup a workspace with the following layout:
  - `rmw_connext_ws`
    - `rti_connext_dds-6.1.2` (containing the whole Pro package as well as the micro package in `rti_connext_dds_micro-3.0.3.28`)
    - `src/rmw_connextdds`
2. put `COLCON_IGNORE` in rmw_connext_ws/rti_connext_dds-6.1.2
3. run `docker run -it -v $PWD:/home -w /home osrf/ros:rolling-desktop in rmw_connext_ws`
4. run `source ./rti_connext_dds-6.1.2/resource/scripts/rtisetenv_x64Linux4gcc7.3.0.bash`
5. run `colcon build`

clalancette, 2023-07-31:
We don't officially support Connext Micro, so we don't usually test it.  For that matter, we don't even have licenses for Connext Micro.  If you have patches to make this compile, we'd be happy to review them.

fgallegosalido, 2026-09-10:
Hi @cwecht . Currently, we are not doing any further development on the Connext Micro RMW. As it is deprecated, I will close this issue.

closed 2026-09-10 by fgallegosalido