ros2/rmw_connextdds #135: On windows (via robostack) can't find connext implementation

tony-p, 2023-11-01:
I'm trying to setup the `rmw_connextdds` on windows (64bit) and during compilation of a CPP package it complains that it can't find the implementation

```sh
 CMake Error at C:/bh_test/.pixi/env/Library/share/rmw_implementation_cmake/cmake/get_default_rmw_implementation.cmake:60 (message):
[4.454s]   Could not find ROS middleware implementation 'rmw_connextdds'.  Choose one
[4.454s]   of the following: rmw_fastrtps_cpp, rmw_fastrtps_dynamic_cpp
```

I am using a conda pre-compiled `rmw_connextdds` version provided by robostack. I installed this using pixi tooling (https://prefix.dev/docs/pixi/overview)

Pixi will set up a virtual environment and automatically runs prescribed activation scripts. A minimal `pixi.toml` to set up the environment with connextdds.

```toml
[project]
name = "connextdds"
version = "0.1.0"
description = "test"
authors = []
channels = ["conda-forge", "robostack-staging"]
platforms = ["win-64"]

[target.win-64.activation]
scripts = ["connext_dds.bat", ".pixi/install/setup.bat"]

[tasks]
build = {cmd="colcon build --merge-install --build-base .pixi/build --install-base .pixi/install --cmake-args -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DPython_EXECUTABLE=$CONDA_PREFIX/python -DPython3_EXECUTABLE=$CONDA_PREFIX/python" }
test = {cmd="colcon test --merge-install --build-base .pixi/build --install-base .pixi/install"}

[dependencies]
# Base
python = "3.10.*"
# ROS
ros-humble-desktop = "0.10.*"
ros-humble-rmw-connextdds = "*"
ros-humble-rmw-connextdds-common = "*"
ros-humble-rti-connext-dds-cmake-module = "*"

# Build Tools
cmake = "*"
pkg-config = "*"
make = "*"
ninja = "*"
colcon-common-extensions = "*"
catkin_tools = "*"

# Platform dependent build chain
[target.win-64.dependencies]
vs2022_win-64 = "19.*"
```

Where the `connextdds.bat` is 
```bat
call "c:\Program Files\rti_connext_dds-6.0.1\resource\scripts\rtisetenv_x64Win64VS2017.bat"
set RMW_IMPLEMENTATION=rmw_connextdds
set CONNEXTDDS_DIR=${NDDSHOME}
```

clalancette, 2023-11-13:
Sorry, I'm not super familiar with Robostack or Pixie.  I'll suggest you open up an issue using the guidelines at https://robostack.github.io/support.html

tony-p, 2023-11-14:
I have asked on the gitter from robostack, but seems nobody there has experience with the connextdds. 

robostack is just a set of precompiled binaries to work in a virtual environment. Pixi is just a tool to manage that virtual environment.

What could be the cause for the implementation not being found? Does `c:\Program Files\rti_connext_dds-6.0.1\resource\scripts\rtisetenv_x64Win64VS2017.bat` expect a system wide path and not limited to a virtual environment? Do certain ROS packages need to be rebuilt? is it sensitive to MSVC version? 

All the environment variables set by `rtisetenv_x64Win64VS2017.bat` are set inside the virtual environment

sloretz, 2023-11-16:
@mjcarroll would you be willing to try this on a windows machine? 🧇