ros2/rmw_connextdds #136: Failing test in CI: test_rclcpp gtest_subscription__rmw_connextdds

clalancette, 2023-11-13:
**System Info**

- _OS_
  - Ubuntu 22.04
- *ROS version and installation type*
  - rolling/source
- *RTI Connext DDS version and installation type*
  - Connext Pro 6.0.1
- *RMW version or commit hash*
  - rolling branch (a6053be96c4cddf4cece95424a68fbb54373e705 as of this writing)

**Bug Description**

If you run the `gtest_subscription__rmw_connextdds` test from the `test_rclcpp` package with no load, it "always" passes.  If also "always" seems to pass in CI.  However, if you put load on the machine, it fails very often, maybe like 75% of the time.

**Expected Behavior**

The test always passes, even with a lot of load on the machine.

**How to Reproduce**

In terminal 1, put a lot of stress on the machine.  In my case:

```
$ stress --cpu 16
```

In terminal 2, run the test:
```
$ colcon test --event-handlers console_direct+ --packages-select test_rclcpp --ctest-args -R gtest_subscription__rmw_connextdds
```

You may have to adjust the amount of stress on the machine, and you may have to run the test a few times, but it should fail fairly quickly.

**Workarounds**

Mark this test as xfail.

**Additional context**

This same test works fine on Fast-DDS (`gtest_subscription__rmw_fastrtps_cpp`) and Cyclone DDS (`gtest_subscription__rmw_cyclonedds_cpp`), even with load on the machine.

This has come up now because we are about to merge in https://github.com/ros2/rclcpp/pull/2142, which seems to exacerbate this problem.  However, I can produce this completely with `rolling` packages as of today, so it is not the fault of that PR.

I did some additional debugging to try to track this down.  When it fails, the executor is waiting for new data to come in via `rmw_wait` via a condition variable: https://github.com/ros2/rmw_connextdds/blob/a6053be96c4cddf4cece95424a68fbb54373e705/rmw_connextdds_common/src/common/rmw_impl_waitset_std.cpp#L579 .  The failure occurs because the condition variable never wakes up and times out.  In turn, this suggests that the condition variable is never triggered.

That condition variable should be triggered when new data comes in via the `on_data_available` callback in the subscriber: https://github.com/ros2/rmw_connextdds/blob/a6053be96c4cddf4cece95424a68fbb54373e705/rmw_connextdds_common/src/common/rmw_impl_waitset_std.cpp#L131 .  I've verified that it is installed to the listener at https://github.com/ros2/rmw_connextdds/blob/a6053be96c4cddf4cece95424a68fbb54373e705/rmw_connextdds_common/src/common/rmw_impl_waitset_std.cpp#L630 .

Finally, I've verified that the publisher side is indeed writing the data out via `DDS_DataWriter_write_untypedI` in message (https://github.com/ros2/rmw_connextdds/blob/a6053be96c4cddf4cece95424a68fbb54373e705/rmw_connextdds_common/src/ndds/dds_api_ndds.cpp#L771) .

So as far as I can tell, those are all of the pieces necessary to get this working, and it *does* work sometimes.  But under load, it seems to fail.  I could use some advice on how to debug this further.

In the meantime, I'm going to propose a PR to mark that particular test as xfail so we can make progress on the other PRs.

fujitatomoya, 2024-04-15:
i did not get to the root cause, https://github.com/ros2/rosbag2/blob/4882d30fc2d1e5b9305e5d46b8460466f9280d27/rosbag2_transport/test/rosbag2_transport/test_burst.cpp#L404-L408 looks like the same issue. when it comes to transient workload, it sometimes miss the subscription data, that leads to the test failure of that test here.

Crola1702, 2024-08-21:
I think this issue is happening consistently on Windows repeated, and it's also flaky on windows debug and rhel repeated.

It's a timeout of this test.

Reference builds:
* https://ci.ros2.org/job/nightly_win_rep/3453/ (consistent)
* https://ci.ros2.org/job/nightly_linux-rhel_repeated/1999/ (flaky)

Test regression:
* [projectroot.gtest_multithreaded__rmw_connextdds](https://ci.ros2.org/job/nightly_win_rep/3453/testReport/junit/(root)/projectroot/gtest_multithreaded__rmw_connextdds/)

Log output:
```
  Start 12: gtest_multithreaded__rmw_connextdds

12: Test command: C:\Python38\python.exe "-u" "C:/ci/ws/install/share/ament_cmake_test/cmake/run_test.py" "C:/ci/ws/build/test_rclcpp/test_results/test_rclcpp/gtest_multithreaded__rmw_connextdds.gtest.xml" "--package-name" "test_rclcpp" "--output-file" "C:/ci/ws/build/test_rclcpp/ament_cmake_gtest/gtest_multithreaded__rmw_connextdds.txt" "--env" "RMW_IMPLEMENTATION=rmw_connextdds" "--append-env" "PATH=C:/ci/ws/build/test_rclcpp/Release" "--command" "C:/ci/ws/build/test_rclcpp/Release/gtest_multithreaded.exe" "--gtest_output=xml:C:/ci/ws/build/test_rclcpp/test_results/test_rclcpp/gtest_multithreaded__rmw_connextdds.gtest.xml"
12: Working Directory: C:/ci/ws/build/test_rclcpp
12: Test timeout computed to be: 90
12: -- run_test.py: extra environment variables:
12:  - RMW_IMPLEMENTATION=rmw_connextdds
12: -- run_test.py: extra environment variables to append:
12:  - PATH+=C:/ci/ws/build/test_rclcpp/Release
12: -- run_test.py: invoking following command in 'C:\ci\ws\build\test_rclcpp':
12:  - C:/ci/ws/build/test_rclcpp/Release/gtest_multithreaded.exe --gtest_output=xml:C:/ci/ws/build/test_rclcpp/test_results/test_rclcpp/gtest_multithreaded__rmw_connextdds.gtest.xml
12: Running main() from C:\ci\ws\install\src\gtest_vendor\src\gtest_main.cc
12: [==========] Running 5 tests from 1 test suite.
12: [----------] Global test environment set-up.
12: [----------] 5 tests from test_multithreaded
12: [ RUN      ] test_multithreaded.multi_consumer_single_producer
    Test #12: gtest_multithreaded__rmw_connextdds .............................***Timeout  90.01 sec
```

Flaky ratio:
job_name|last_fail|first_fail|build_count|failure_count|failure_percentage
-- | -- | -- | -- | -- | --
nightly_win_rep|2024-08-21|2024-08-06|13|13|100.0
nightly_win_deb|2024-08-20|2024-08-20|14|1|7.14
nightly_linux-rhel_repeated|2024-08-18|2024-08-14|16|2|12.5
nightly_linux_coverage|2024-08-07|2024-08-07|15|1|6.67

WDYT @clalancette?

Crola1702, 2024-10-09:
Reference build: https://ci.ros2.org/job/nightly_win_rep/3504/

Failing test: `test_multithreaded.multi_access_publisher` with ConnextDDS

<details><summary>Log output</summary>

```
[ RUN      ] test_multithreaded.multi_access_publisher
[WARN] [1728461640.564957000] [rclcpp]: logging was initialized more than once
Waited 0 microseconds for the subscriber to connect to topic 'multi_access_publisher'
Publishing message 0
Publishing message 1
Publishing message 2
Publishing message 3
Publishing message 4
Publishing message 5
Publishing message 6
Publishing message 7
Publishing message 8
Publishing message 9
Publishing message 10
Publishing message 11
Subscription callback 0
callback() 0 with message data 7
Publishing message 12
Subscription callback 1
callback() 1 with message data 8
Subscription callback 2
callback() 2 with message data 9
Publishing message 13
Subscription callback 3
callback() 3 with message data 10
Publishing message 14
Subscription callback 4
callback() 4 with message data 11
Subscription callback 5
callback() 5 with message data 12
Subscription callback 6
callback() 6 with message data 13
Publishing message 15
Subscription callback 7
callback() 7 with message data 14
Publishing message 16
Subscription callback 8
callback() 8 with message data 15
Publishing message 17
Subscription callback 9
callback() 9 with message data 16
Publishing message 18
Subscription callback 10
callback() 10 with message data 17
Subscription callback 11
callback() 11 with message data 18
Publishing message 19
Subscription callback 12
callback() 12 with message data 19
C:\ci\ws\src\ros2\system_tests\test_rclcpp\test\test_multithreaded.cpp(421): error: Expected equality of these values:
  num_messages
    Which is: 20
  subscription_counter
    Which is: 13

[  FAILED  ] test_multithreaded.multi_access_publisher (5521 ms)
```

</details>

I see this error is happening mostly on windows. These started failing a lot more since testing parallelization in Feb 25th. Check:
* First windows debug failure this year: https://ci.ros2.org/view/nightly/job/nightly_win_deb/3014/
* First windows repeated failure this year: https://ci.ros2.org/view/nightly/job/nightly_win_rep/3275/

Flaky ratio in the last 15 days:
job_name|last_fail|first_fail|build_count|failure_count|failure_percentage
-- | -- | -- | -- | -- | --
nightly_win_rep|2024-10-09|2024-09-25|10|10|100.0
nightly_win_deb|2024-10-09|2024-09-27|14|7|50.0
nightly_linux_repeated|2024-10-05|2024-10-05|13|1|7.69
nightly_win_rel|2024-09-28|2024-09-28|13|1|7.69

<details><summary>Flaky ratio since paralelize testing (277 days in oct 09):</summary>

job_name|last_fail|first_fail|build_count|failure_count|failure_percentage
-- | -- | -- | -- | -- | --
nightly_win_rep|2024-10-09|2024-02-25|229|167|72.93
nightly_win_deb|2024-10-09|2024-02-25|209|86|41.15
nightly_linux_repeated|2024-10-05|2024-04-09|246|7|2.85
nightly_win_rel|2024-09-28|2024-03-04|262|11|4.2
nightly_linux-rhel_repeated|2024-08-18|2024-03-15|265|7|2.64
nightly_linux_coverage|2024-08-07|2024-03-29|224|12|5.36
nightly_linux_debug|2024-04-07|2024-04-01|247|5|2.02

</details>

fujitatomoya, 2025-02-06:
@lobolanja @fgallegosalido could you please take a look at this? this has been CI failure for a long time.

fgallegosalido, 2025-02-06:
@fujitatomoya we'll take look. We have quite a big backlog right now, but we'll keep track of the issues as they appear and look into them as soon as possible.

Crola1702, 2025-03-04:
This is still happening for Windows:

Reference build:
* https://ci.ros2.org/job/nightly_win_rep/3657/

Test regression:
* [test_rclcpp.gtest_multithreaded__rmw_connextdds.gtest.missing_result](https://ci.ros2.org/job/nightly_win_rep/3657/testReport/junit/(root)/test_rclcpp/gtest_multithreaded__rmw_connextdds_gtest_missing_result/)

Log output:
```
      Start 12: gtest_multithreaded__rmw_connextdds

12: Test command: C:\Python38\python.exe "-u" "C:/ci/ws/install/share/ament_cmake_test/cmake/run_test.py" "C:/ci/ws/build/test_rclcpp/test_results/test_rclcpp/gtest_multithreaded__rmw_connextdds.gtest.xml" "--package-name" "test_rclcpp" "--output-file" "C:/ci/ws/build/test_rclcpp/ament_cmake_gtest/gtest_multithreaded__rmw_connextdds.txt" "--env" "RMW_IMPLEMENTATION=rmw_connextdds" "--append-env" "PATH=C:/ci/ws/build/test_rclcpp/Release" "--command" "C:/ci/ws/build/test_rclcpp/Release/gtest_multithreaded.exe" "--gtest_output=xml:C:/ci/ws/build/test_rclcpp/test_results/test_rclcpp/gtest_multithreaded__rmw_connextdds.gtest.xml"
12: Working Directory: C:/ci/ws/build/test_rclcpp
12: Test timeout computed to be: 90
12: -- run_test.py: extra environment variables:
12:  - RMW_IMPLEMENTATION=rmw_connextdds
12: -- run_test.py: extra environment variables to append:
12:  - PATH+=C:/ci/ws/build/test_rclcpp/Release
12: -- run_test.py: invoking following command in 'C:\ci\ws\build\test_rclcpp':
12:  - C:/ci/ws/build/test_rclcpp/Release/gtest_multithreaded.exe --gtest_output=xml:C:/ci/ws/build/test_rclcpp/test_results/test_rclcpp/gtest_multithreaded__rmw_connextdds.gtest.xml
12: Running main() from C:\ci\ws\install\src\gtest_vendor\src\gtest_main.cc
12: [==========] Running 5 tests from 1 test suite.
12: [----------] Global test environment set-up.
12: [----------] 5 tests from test_multithreaded
12: [ RUN      ] test_multithreaded.multi_consumer_single_producer
    Test #12: gtest_multithreaded__rmw_connextdds .............................***Timeout  90.00 sec
```

fgallegosalido, 2025-04-11:
I wanted to provide an update on this issue. I was able to narrow the failure scenario to just the following test code, which is almost 100% reproducible with `stress --cpu 16` on my machine.:

```cpp
TEST_F(test_subscription, subscription_and_spinning_bug)
{
  auto node = rclcpp::Node::make_shared("test_subscription");

  const auto topic = std::string("test_subscription");

  auto publisher = node->create_publisher<test_rclcpp::msg::UInt32>(topic, rclcpp::QoS(10).transient_local().reliable().keep_all());

  rclcpp::executors::SingleThreadedExecutor executor;
  executor.add_node(node);

  std::promise<void> sub_called;
  std::shared_future<void> sub_called_future(sub_called.get_future());

  auto subscriber = node->create_subscription<test_rclcpp::msg::UInt32>(
    topic,
    10,
    [&sub_called](const test_rclcpp::msg::UInt32::ConstSharedPtr) -> void
    {
      sub_called.set_value();
    });

  test_rclcpp::wait_for_subscriber(node, topic);

  publisher->publish(test_rclcpp::build<test_rclcpp::msg::UInt32>().data(101));

  const auto future_ret = executor.spin_until_future_complete(sub_called_future, 100s);
  EXPECT_EQ(rclcpp::FutureReturnCode::SUCCESS, future_ret);
}
```

After some deep debugging, we found out that the reason the `on_data_available` callback was not waking up is because the writer isn't even putting the sample on the cable (we confirmed this with Wireshark). We are still trying to figure out why that might be the issue, as even changing asynchronous publishing mode to synchronous mode still makes the test fail.

I'll keep you posted on what we find.

Crola1702, 2025-06-10:
Friendly ping @fgallegosalido