ros2/rmw_connextdds #155: :farmer: rclcpp tests failing due to rmw callbacks `rmw_client_set_on_new_{response,request}_callback` not implemented in Humble ci and Iron ci

Crola1702, 2024-07-17:
- Ubuntu jammy and noble
- Humble and Iron source

**How to Reproduce**
1. Run a build in Hci_nightly_connext or Ici_nightly_connext
2. See tests failures

**Bug Description**

Reference builds:
* [Humble#254](https://build.ros2.org/job/Hci__nightly-connext_ubuntu_jammy_amd64/254/)
* [Iron#147](https://build.ros2.org/job/Ici__nightly-connext_ubuntu_jammy_amd64/147/)

Failing tests:

`rmw_client_set_on_new_response_callback not implemented`:
* [projectroot.test.rclcpp.test_client](https://build.ros2.org/job/Hci__nightly-connext_ubuntu_jammy_amd64/254/testReport/junit/projectroot.test/rclcpp/test_client/)
* [rclcpp.TestClient.on_new_response_callback](https://build.ros2.org/job/Hci__nightly-connext_ubuntu_jammy_amd64/254/testReport/junit/rclcpp/TestClient/on_new_response_callback/)

`rmw_event_set_callback not implemented`:
* [projectroot.test.rclcpp.test_qos_event](https://build.ros2.org/job/Hci__nightly-connext_ubuntu_jammy_amd64/254/testReport/junit/projectroot.test/rclcpp/test_qos_event/)
* [rclcpp.TestQosEvent.test_on_new_event_callback](https://build.ros2.org/job/Hci__nightly-connext_ubuntu_jammy_amd64/254/testReport/junit/rclcpp/TestQosEvent/test_on_new_event_callback/)
* [rclcpp.TestQosEvent.test_invalid_on_new_event_callback](https://build.ros2.org/job/Hci__nightly-connext_ubuntu_jammy_amd64/254/testReport/junit/rclcpp/TestQosEvent/test_invalid_on_new_event_callback/)

`rmw_service_set_on_new_request_callback not implemented`:
* [projectroot.test.rclcpp.test_service](https://build.ros2.org/job/Hci__nightly-connext_ubuntu_jammy_amd64/254/testReport/junit/projectroot.test/rclcpp/test_service/)
* [rclcpp.TestService.on_new_request_callback](https://build.ros2.org/job/Hci__nightly-connext_ubuntu_jammy_amd64/254/testReport/junit/rclcpp/TestService/on_new_request_callback/)

`rmw_subscription_set_on_new_message_callback`:
* [projectroot.test.rclcpp.test_subscription](https://build.ros2.org/job/Hci__nightly-connext_ubuntu_jammy_amd64/254/testReport/junit/projectroot.test/rclcpp/test_subscription/)
* [rclcpp.TestSubscription.on_new_message_callback](https://build.ros2.org/job/Hci__nightly-connext_ubuntu_jammy_amd64/254/testReport/junit/rclcpp/TestSubscription/on_new_message_callback/)
 > **Note**: rmw_api_connextdds_subscription_get_network_flow_endpoints: not implemented is also showing here

Full log output:

```
unknown file: Failure
C++ exception with description "failed to set the on new response callback for client: rmw_client_set_on_new_response_callback not implemented, at /tmp/ws/src/ros2/rmw_connextdds/rmw_connextdds_common/src/common/rmw_listener.cpp:58" thrown in the test body.

...

unknown file: Failure
C++ exception with description "failed to set the on new message callback for QOS Event: rmw_event_set_callback not implemented, at /tmp/ws/src/ros2/rmw_connextdds/rmw_connextdds_common/src/common/rmw_listener.cpp:29" thrown in the test body.

...

unknown file: Failure
C++ exception with description "failed to set the on new request callback for service: rmw_service_set_on_new_request_callback not implemented, at /tmp/ws/src/ros2/rmw_connextdds/rmw_connextdds_common/src/common/rmw_listener.cpp:45" thrown in the test body.

...

unknown file: Failure
C++ exception with description "failed to set the on new message callback for subscription: rmw_subscription_set_on_new_message_callback not implemented, at /tmp/ws/src/ros2/rmw_connextdds/rmw_connextdds_common/src/common/rmw_listener.cpp:74" thrown in the test body.
```

Crola1702, 2024-07-30:
@asorbini do you think we can have a fix for this one?

fujitatomoya, 2024-07-30:
@trubio-rti friendly ping.

trubio-rti, 2024-07-31:
@Crola1702 @fujitatomoya I've been on vacation the past two weeks and didn't get any notifications for this. Please add both of us as assignees from now on so I can get an email as well :)

claraberendsen, 2024-10-15:
Closing since this has been resolved for Humble and will not be forward ported to Iron since it's close to it's EOL.

PR #157 (2024-08-20): Backport rmw callbacks implementation to Humble [ros2-73]
This fixes #155. I've also sneakily backported the fix to allow Humble to build with Connext 7.3.0.

`ci_linux`: [image]
`ci_linux_rhel`: [image]
`ci_windows`: [image]
closed 2024-10-15 by claraberendsen