ros2/rmw_connextdds #159: RMW_CONNEXT_ENDPOINT_QOS_OVERRIDE_POLICY clarification

scn-nmsc, 2024-08-22:
**System Info**

- Ubuntu 20.04
- jazzy/binary
- Pro 6.0.1/rtipkg

**Bug Description**

When using RMW_CONNEXT_ENDPOINT_QOS_OVERRIDE_POLICY="dds_topics:rt/my_topic|rt/my_other_topic" and a USER_QOS_PROFILE.xml in the cwd with a topic filter for "rt/my_topic" I would assume that the QoS settings are determined by the profile specified in xml.

In my example I specify a profile with best_effort and deadline for "rt/my_topic" in my xml file.
The subscription by the ros node will however end up with:

1. using the ros qos_profile_sensor_data
-  best_effort
-  deadline (as specified in xml)

2. using the e.g. ros QosProfile(depth=1)
-  reliable
-  deadline (as specified in xml)

Meaning the deadline is always set as defined in the xml file, but reliability is determined by the profile specified in the ros application.

**Expected Behavior**

Profiles in xml determine the QoS settings for topics specified by RMW_CONNEXT_ENDPOINT_QOS_OVERRIDE_POLICY="dds_topics:rt/my_topic|rt/my_other_topic" and topic filter in USER_QOS_PROFILES.xml file.

Alternative solution: a more specific README on how the different QoS settings are determined. For now I am confused by the README and the observed behavior.

gabriel-st-angel-aerospace, 2025-04-23:
Also interested in the solution to this

gabriel-st-angel-aerospace, 2025-04-23:
@asorbini

gabriel-st-angel-aerospace, 2025-04-24:
@fgallegosalido I have one query that I tagged you in, it should be quick for you guys to answer.  thanks for your help