ros2/rmw_connextdds #170: 🧑‍🌾 Rolling CI warning: `format ‘%s’ expects a matching ‘char*’ `

Crola1702, 2025-02-26:
**System Info**

- _OS_
  - [Ubuntu 20.04
- *ROS version and installation type*
  - Rolling
- *RTI Connext DDS version and installation type*
  - CI Install

**Bug Description**

Reference builds:
- https://ci.ros2.org/job/nightly_linux_repeated/3746/gcc/
- https://ci.ros2.org/job/nightly_linux_coverage/2690/gcc/
- https://build.ros2.org/job/Rci__nightly-release_ubuntu_noble_amd64/302/gcc/
- https://build.ros2.org/job/Rci__nightly-debug_ubuntu_noble_amd64/300/gcc/
- https://build.ros2.org/job/Rci__nightly-connext_ubuntu_noble_amd64/309/gcc/new/

Warning:
- [rmw_context.cpp:473](https://ci.ros2.org/job/nightly_linux_repeated/3746/gcc/source.3af33c60-fb21-4440-a5c0-b9a8460f5580/#473): `format ‘%s’ expects a matching ‘char*’ argument [-Wformat=] 473 | "failed to assert DDS property: '%s' = '%s'",`

```
In file included from /home/jenkins-agent/workspace/nightly_linux_repeated/ws/src/ros2/rmw_connextdds/rmw_connextdds_common/include/rmw_connextdds/log.hpp:20,
                 from /home/jenkins-agent/workspace/nightly_linux_repeated/ws/src/ros2/rmw_connextdds/rmw_connextdds_common/include/rmw_connextdds/context.hpp:24,
                 from /home/jenkins-agent/workspace/nightly_linux_repeated/ws/src/ros2/rmw_connextdds/rmw_connextdds_common/include/rmw_connextdds/rmw_impl.hpp:25,
                 from /home/jenkins-agent/workspace/nightly_linux_repeated/ws/src/ros2/rmw_connextdds/rmw_connextdds_common/src/common/rmw_context.cpp:26:
/home/jenkins-agent/workspace/nightly_linux_repeated/ws/src/ros2/rmw_connextdds/rmw_connextdds_common/src/common/rmw_context.cpp: In member function ‘rmw_ret_t rmw_context_impl_s::configure_security(DDS_DomainParticipantQos*)’:
/home/jenkins-agent/workspace/nightly_linux_repeated/ws/src/ros2/rmw_connextdds/rmw_connextdds_common/src/common/rmw_context.cpp:473:7: warning: format ‘%s’ expects a matching ‘char*’ argument [-Wformat=]
  473 |       "failed to assert DDS property: '%s' = '%s'",
/home/jenkins-agent/workspace/nightly_linux_repeated/ws/src/ros2/rmw_connextdds/rmw_connextdds_common/include/rmw_connextdds/log.hpp:195:3: note: in expansion of macro ‘RMW_CONNEXT_LOG_ERROR_A’
  195 |   RMW_CONNEXT_LOG_ERROR_A(fmt_, __VA_ARGS__) \
      |   ^~~~~~~~~~~~~~~~~~~~~~~
/home/jenkins-agent/workspace/nightly_linux_repeated/ws/src/ros2/rmw_connextdds/rmw_connextdds_common/src/common/rmw_context.cpp:472:5: note: in expansion of macro ‘RMW_CONNEXT_LOG_ERROR_A_SET’
  472 |     RMW_CONNEXT_LOG_ERROR_A_SET(
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/jenkins-agent/workspace/nightly_linux_repeated/ws/install/rmw/include/rmw/rmw/error_handling.h:31,
                 from /home/jenkins-agent/workspace/nightly_linux_repeated/ws/src/ros2/rmw_connextdds/rmw_connextdds_common/src/common/rmw_context.cpp:24:
/home/jenkins-agent/workspace/nightly_linux_repeated/ws/src/ros2/rmw_connextdds/rmw_connextdds_common/src/common/rmw_context.cpp:473:7: warning: format ‘%s’ expects a matching ‘char*’ argument [-Wformat=]
  473 |       "failed to assert DDS property: '%s' = '%s'",
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/jenkins-agent/workspace/nightly_linux_repeated/ws/src/ros2/rmw_connextdds/rmw_connextdds_common/src/common/rmw_context.cpp:472:5: note: in expansion of macro ‘RMW_CONNEXT_LOG_ERROR_A_SET’
  472 |     RMW_CONNEXT_LOG_ERROR_A_SET(
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/jenkins-agent/workspace/nightly_linux_repeated/ws/src/ros2/rmw_connextdds/rmw_connextdds_common/src/common/rmw_context.cpp:473:48: note: format string is defined here
  473 |       "failed to assert DDS property: '%s' = '%s'",
      |                                               ~^
      |                                                |
      |                                                char*
```

fgallegosalido, 2025-02-26:
This seems to come from https://github.com/ros2/rmw_connextdds/pull/167. The fix is just adding to the line 474 in the file `rmw_context.hpp` the string `DDS_SECURITY_PRIVATE_KEY_PROPERTY`. This is consistent with the rest of the changes,

fujitatomoya, 2025-02-26:
@fgallegosalido thanks, here is the fix https://github.com/ros2/rmw_connextdds/pull/171.

closed 2025-02-27 by ahcorde