ros2/rmw_connextdds #190: Debian binary installation is missing `librmw_connextdds.so`

fujitatomoya, 2025-05-02:
**System Info**

- _OS_
  - Ubuntu 24.04
- *ROS version and installation type*
  - Kilted Kaiju
- *RTI Connext DDS version and installation type*
  - Debian v7.3.0
- *RMW version or commit hash*
  - N.A

**Bug Description**

https://github.com/ros2/kilted_tutorial_party/issues/282

**Expected Behavior**

Installation and demonstration should succeed.

**How to Reproduce**

https://docs.ros.org/en/kilted/Installation/RMW-Implementations/DDS-Implementations/Working-with-RTI-Connext-DDS.html

**Workarounds**

None, this blocks kilted user to try rmw_connextdds

**Additional context**

[Any additional relevant information about the problem]

fujitatomoya, 2025-05-02:
@fgallegosalido @lobolanja can you take a look at this? i think this could be priority.

slowrunner, 2025-05-02:
Indeed this is blocking 
https://github.com/ros2/kilted_tutorial_party/issues/281
https://github.com/ros2/kilted_tutorial_party/issues/283

slowrunner, 2025-05-02:
The ros connext binary for arm64 appears to be for bullseye, but Ubuntu24.04 is based on "bookworm" - 
```
rti-connext-dds-7.3.0-ros                         7.3.0.0-5-4bullseye                     arm64        RTI Connext 7.3.0 for ROS 
```

fgallegosalido, 2025-05-05:
@slowrunner new binaries (at least in amd64) now include the libraries. Give a try and let us know if it works

slowrunner, 2025-05-05:
> Give a try and let us know if it works

Yes, works on arm64 kilted (Raspberry Pi 4 / Ubuntu 24.04 / ROS 2 Kilted Kaiju)

```
ubuntu@kilteddave:~/KiltedDave/ros2ws$ export RMW_IMPLEMENTATION=rmw_connextdds
ubuntu@kilteddave:~/KiltedDave/ros2ws$ ros2 run demo_nodes_cpp listener
RTI Connext DDS Non-commercial license is for academic, research, evaluation and personal use only. USE FOR COMMERCIAL PURPOSES IS PROHIBITED. See RTI_LICENSE.TXT for terms. Download free tools at rti.com/ncl. License issued to Non-Commercial User license@rti.com For non-production use only.
Expires on 00-jan-00
Please contact support@rti.com with any questions or comments.
[INFO] [1746445247.316045391] [listener]: I heard: [Hello World: 0]
[INFO] [1746445248.223488148] [listener]: I heard: [Hello World: 1]
[INFO] [1746445249.224182061] [listener]: I heard: [Hello World: 2]
[INFO] [1746445250.224070586] [listener]: I heard: [Hello World: 3]
[INFO] [1746445251.224210609] [listener]: I heard: [Hello World: 4]
^C[INFO] [1746445254.307434904] [rclcpp]: signal_handler(signum=2)
[ros2run]: Received signal:  Interrupt
[INFO] [1746445254.308240216] [rclcpp]: signal_handler(signum=2)
ubuntu@kilteddave:~/KiltedDave/ros2ws$ 

```

fujitatomoya, 2025-05-05:
@fgallegosalido i confirmed the latest binary.

```console
root@tomoyafujita:~# source /opt/ros/kilted/setup.bash
root@tomoyafujita:~# RMW_IMPLEMENTATION=rmw_connextdds ros2 run demo_nodes_cpp talker
RTI Connext DDS Non-commercial license is for academic, research, evaluation and personal use only. USE FOR COMMERCIAL PURPOSES IS PROHIBITED. See RTI_LICENSE.TXT for terms. Download free tools at rti.com/ncl. License issued to Non-Commercial User license@rti.com For non-production use only.
Expires on 00-jan-00
Please contact support@rti.com with any questions or comments.
[INFO] [1746458916.019775906] [talker]: Publishing: 'Hello World: 1'
[INFO] [1746458917.019745979] [talker]: Publishing: 'Hello World: 2'
^C[INFO] [1746458917.240010324] [rclcpp]: signal_handler(signum=2)
[ros2run]: Received signal:  Interrupt
[INFO] [1746458917.240146307] [rclcpp]: signal_handler(signum=2)
```

closed 2025-05-05 by fujitatomoya