ros2/rmw_connextdds #191: `ros2 param list` does not work

fujitatomoya, 2025-05-13:
**System Info**

- _OS_
  - Ubuntu 24.04]
  - *ROS version and installation type*
  - kilted/binary, rolling/source
- *RTI Connext DDS version and installation type*
  - 7.3.0.0-5-4bullseye
- *RMW version or commit hash*
  - 1.1.0-3noble.20250502.195232

**Bug Description**

```console
root@tomoyafujita:/# ros2 daemon status
The daemon is not running
root@tomoyafujita:/# ros2 param list
RTI Connext DDS Non-commercial license is for academic, research, evaluation and personal use only. USE FOR COMMERCIAL PURPOSES IS PROHIBITED. See RTI_LICENSE.TXT for terms. Download free tools at rti.com/ncl. License issued to Non-Commercial User license@rti.com For non-production use only.
Expires on 00-jan-00
Please contact support@rti.com with any questions or comments.
RTI Connext DDS Non-commercial license is for academic, research, evaluation and personal use only. USE FOR COMMERCIAL PURPOSES IS PROHIBITED. See RTI_LICENSE.TXT for terms. Download free tools at rti.com/ncl. License issued to Non-Commercial User license@rti.com For non-production use only.
Expires on 00-jan-00
Please contact support@rti.com with any questions or comments.
...
```

and no response at all.

**Expected Behavior**

should be able to list the parameters.

**How to Reproduce**

Described above. this can be reproducible with rolling source build. i am guessing that something wrong with `rti-connext-dds-7.3.0-ros`, because i have never seen this behavior with `rti-connext-dds-6.0.1`.

**Workarounds**

None

**Additional context**

Originally found with https://github.com/ros2/kilted_tutorial_party/issues/412

fgallegosalido, 2025-05-13:
Looking into it

fgallegosalido, 2025-05-15:
Is this also happening in Jazzy with Connext 6.0.1? It seems that I can reproduce this in Jazzy too, but maybe I'm doing something wrong.

fujitatomoya, 2025-05-15:
as far as i checked, jazzy does not have problem like this.

fgallegosalido, 2025-05-22:
I haven't had much time to look into this. I will try to find the issue before the release. Otherwise I would try to introduce the fix when I find it and backport it to Kilted