ros2/rmw_connextdds #192: `ros2 param list` fails on Windows 10

zamantousif, 2025-05-14:
**System Info**

- _OS_
  - [Windows 10]
- *ROS version and installation type*
  - [rolling/source]
- *RTI Connext DDS version and installation type*
  - [7.5.0]
- *RMW version or commit hash*
  - [`f8d42bd7a4fca543b8c0ffe03e24bb61b2944564 `]

**Bug Description**

`ros2 param list` fails with the following error message whereas other ros2cli commands such as `ros2 param list`, `ros2 node list` and `ros2 service list` seem to work.

[image]

**Expected Behavior**

`ros2 param list` should return without the above message or error.

**How to Reproduce**

**Additional context**

Found the issue while working on https://github.com/ros2/kilted_tutorial_party/issues/389

fujitatomoya, 2025-05-14:
@zamantousif thanks for creating issue,

i think this is duplication of https://github.com/ros2/rmw_connextdds/issues/191.

zamantousif, 2025-05-14:
@fujitatomoya I did check the bug issue opened by you with RTI 7.3.0 however, I am using the latest RTI Connext 7.5.0. Is this issue still considered duplicate? 🤔 If so, I could close it and mark duplicate. Please let me know your opinion.😄

fujitatomoya, 2025-05-14:
maybe we can keep this open until we confirm that this is the same root cause.

zamantousif, 2025-05-20:
Tried with RTI 7.3.0 and ros2 param list does work

[image], also relevant for #193

zamantousif, 2025-05-21:
Closing this ticket as currently supported RMW

closed 2025-05-21 by zamantousif