ros2/rmw_connextdds #193: `rqt` plugins and basic `ros2 node launch` or `ros2 run` do not work with RTI 7.5.0

zamantousif, 2025-05-17:
**System Info**

- _OS_
  - [Windows 10]
- *ROS version and installation type*
  - [kilted/binary, rolling/source]
- *RTI Connext DDS version and installation type*
  - [7.5.0]
- *RMW version or commit hash*
  - [f8d42bd7a4fca543b8c0ffe03e24bb61b2944564 ]

**Bug Description**

Tried basic ros2 checks with RTI 7.5.0 evaluation version on Windows 10 machine and it fails.
1. `rqt` opens but with some prints and crashes as soon as a plugin is loaded via `rqt -> Plugins -> Visualization -> Plot` causing the rqt window to immediately exit. There is no exit code printed however.
```
(pixi_ros2_rolling) C:\dev\ros2_connextdds>rqt                                                                       
RosPluginProvider._parse_plugin_xml() plugin file "C:\dev\kilted\install\share\rqt_gui_cpp/plugin.xml" in package "rqt_gui_cpp" not found                                                                                                 
RosPluginProvider._parse_plugin_xml() plugin file "C:\dev\kilted\install\share\rqt_gui_cpp/plugin.xml" in package "rqt_gui_cpp" not found
RTI Connext DDS Evaluation License issued to <email> For non-production use only.      Expires on 13-jun-2025                                                                                               Please contact support@rti.com with any questions or comments.  
```
2. Launching nodes does not work, simple `talker`, `listener` all lead to runtime errors and process exit with failure.
[image]

```
(pixi_ros2_rolling) C:\dev\ros2_connextdds>ros2 run turtlesim turtle_teleop_key                                      
RTI Connext DDS Evaluation License issued to <email> For non-production use only.      Expires on 13-jun-2025                                                                                               Please contact support@rti.com with any questions or comments.                                                       
[ros2run]: Process exited with failure 3221225477                                        
```

**Expected Behavior**
1. rqt shall open without complaining about missing plugins. Adding plugins to rqt window should work smoothly without any runtime failures. If there is a failure, the appropriate exit code should be returned.
2. Launching a node and running simple talker/listener examples from ros2 documentation should work fine with rmw_connextdds as the RMW_IMPLEMENTATION without any runtime failures.

**How to Reproduce**
1. Build ros2 rolling from source
2. Install RTI 7.5.0 with evaluation license
3. Install rmw_connextdds from this repository in an overlay workspace and source ros2, rmw_connextdds workspace, set RMW_IMPLEMENTATION=rmw_connextdds
4. Run simple talker/listener example from ros2 doc or launch rqt or simply launch a node with launch file

**Additional context**
I have noticed some commands within ros2cli to work however, the above do not seem to work. This ticket is only related to the above commands. It is possible that there are other issues too.

christophebedard, 2025-05-17:
It's important to note that we currently support RTI Connext DDS version 7.3.0, not 7.5.0. Similar to #192, please try with version 7.3.0 and see if you can reproduce the same issue. If not, we can probably accept fixes that will make version 7.5.0 work, but again it's not officially supported.

zamantousif, 2025-05-20:
Okay, I tried with RTI 7.3.0 and rqt, ros2 launch and ros2 run seem to work fine. Please find my latest comments related to this [#280](https://github.com/ros2/kilted_tutorial_party/issues/280#issuecomment-2895803303), [#284](https://github.com/ros2/kilted_tutorial_party/issues/284#issuecomment-2895818498) , [#440](https://github.com/ros2/kilted_tutorial_party/issues/440#issuecomment-2895869667)

zamantousif, 2025-05-20:
@christophebedard how do we proceed with this ticket and #192 now that it works with 7.3.0?

christophebedard, 2025-05-20:
I would close issues related to 7.5.0 (this one and #192), since it's not officially supported. That being said, we can probably accept PRs that make it work with RTI Connext DDS 7.5.0 as long as they don't break compatibility with version 7.3.0.

zamantousif, 2025-05-21:
Sounds fair, I will close the tickets. 
@christophebedard could you point me to where I could start to do the fixes if you have some ideas? I am not sure if this is a ROS2 issue or an issue with rmw_connextdds 7.5.0 😅

fgallegosalido, 2025-05-21:
@zamantousif we are starting to work on supporting Connext 7.5.0 for Rolling. Something must have changed between versions that the RMW is not really enjoying. I'll open an issue later to track the progress there. Thanks for all the testing!

closed 2025-05-21 by zamantousif